nodemcu/nodemcu-firmware #1515: gcc warning for spiffs_gc.c

devsaurus, 2016-09-28:
### Expected behavior
Compilation is clean.

### Actual behavior
Compile-time warning emitted, is it relevant?
```
make[2]: Entering directory 'tools/spiffsimg'
gcc -g -Wall -Wextra -Wno-unused-parameter -Wno-unused-function -I. -I../../app/spiffs -I../../app/include -DNODEMCU_SPIFFS_NO_INCLUDE --include spiffs_typedefs.h main.c ../../app/spiffs/spiffs_cache.c ../../app/spiffs/spiffs_check.c ../../app/spiffs/spiffs_gc.c ../../app/spiffs/spiffs_hydrogen.c ../../app/spiffs/spiffs_nucleus.c  -o spiffsimg
../../app/spiffs/spiffs_gc.c: In function ‘spiffs_gc_find_candidate’:
../../app/spiffs/spiffs_gc.c:258:27: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
   cand_scores = (s32_t*)(((ptrdiff_t)cand_scores + sizeof(ptrdiff_t) - 1) & ~(sizeof(ptrdiff_t) - 1));
                           ^
../../app/spiffs/spiffs_gc.c:258:17: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
   cand_scores = (s32_t*)(((ptrdiff_t)cand_scores + sizeof(ptrdiff_t) - 1) & ~(sizeof(ptrdiff_t) - 1));
                 ^
```

### NodeMCU version
Current `dev`.

pjsg, 2016-09-28:
what platform is this on? I'm suspecting some sort of 32bit/64bit issue. It is probably critical that spiffsimg be built as a 32-bit executable (to match the ESP8266)....

devsaurus, 2016-09-28:
It's a Linux 32bit box:

```
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/i586-suse-linux/4.8/lto-wrapper
Target: i586-suse-linux
Configured with: ../configure --prefix=/usr --infodir=/usr/share/info --mandir=/usr/share/man --libdir=/usr/lib --libexecdir=/usr/lib --enable-languages=c,c++,objc,fortran,obj-c++,java,ada --enable-checking=release --with-gxx-include-dir=/usr/include/c++/4.8 --enable-ssp --disable-libssp --disable-plugin --with-bugurl=http://bugs.opensuse.org/ --with-pkgversion='SUSE Linux' --disable-libgcj --disable-libmudflap --with-slibdir=/lib --with-system-zlib --enable-__cxa_atexit --enable-libstdcxx-allocator=new --disable-libstdcxx-pch --enable-version-specific-runtime-libs --enable-linker-build-id --enable-linux-futex --program-suffix=-4.8 --without-system-libunwind --with-arch-32=i586 --with-tune=generic --build=i586-suse-linux --host=i586-suse-linux
Thread model: posix
gcc version 4.8.3 20140627 [gcc-4_8-branch revision 212064] (SUSE Linux)
```

No warnings on a 64bit system...

jmattsson, 2016-09-29:
Looks like it should be `intptr_t` for starters - that's the safe type for carrying pointers as integers.

The cheap & easy way out might be to force -m32 though.

pjsg, 2016-09-29:
It turns out that I don't have the 32 bit libraries on my platform (at least, adding -m32 causes my build to fail). However, since it generates the right spiffs data file, I suspect that the code works when compiled with a different pointer size. I'll dig in some more, and file a ticket with the spiffs project (and possibly a PR). 

I only get the warning if I try and compile with -m32.....

pjsg, 2016-10-01:
The upstream code has now been fixed. I'll update our version.....

marcelstoer, 2017-04-30:
@pjsg there hasn't been a PR in the meantime, has there (one w/o a back-link to this)?

PR #1949 (2017-05-21): Update to current version of SPIFFS
Fixes #1515.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).

I ran the update_spiffs.sh command to fetch the latest version of SPIFFS. I had to update the spiffs_config.h for some missing defines.

This builds on my system. Hopefully someone with the original problem can verify this.
closed 2017-05-21 by marcelstoer