nodemcu/nodemcu-firmware #1517: UART input during initial fs_format prevents NodeMCU from starting up

jimparis, 2016-09-29:
On an erased chip, NodeMCU starts with "Formatting file system. Please wait...".  This takes a while, 37 seconds on my chip.  If I send 8 or more characters via the UART during this time, it fails to ever get to a Lua prompt.

This is a problem because I have another device trying to detect when Lua is ready by repeatedly sending commands.

I tracked this down, although I am unfamiliar with both NodeMCU and the ESP8266, so I don't know the best solution.  Basically, the UART RX interrupt is calling `task_post` every time a new byte comes in, but since we're still in `user_init` during `fs_format`, those tasks are not getting processed.  The task queue (defined in `task/task.c` with length 8) fills up.  After `fs_format` returns, the final `task_post_low(task_get_id(start_lua,'s')` gets dropped because the queue is already full, and so Lua never starts.

### Steps
* Fully erase a chip
* Flash NodeMCU and `esp_init_data_default.bin`
* Boot the ESP2866
* At the "Formatting file system. Please wait..." prompt, send 10 characters via UART.
* NodeMCU will never start.

### Expected behavior
Sending data during "Formatting file system..." is OK

### Actual behavior
Sending data during "Formatting file system..." prevents Lua startup

### NodeMCU version
8e48483c825dea9c12b37a4db3d034fccbcba0bf

### Hardware
ESP-WROOM-02

jmattsson, 2016-10-02:
Nice bit of debugging, thanks!

I'll prepare a fix for this, stay tuned...

closed 2016-10-08 by marcelstoer

eyaleb, 2016-10-18:
An aside: I found that for a small project that fits in 512KB, setting, in `app/include/user_config.h`
``` c
#define FLASH_512K
...
// #define FLASH_AUTOSIZE
```
makes flash access much faster and a format takes only 1-2 seconds.

PR #1522 (2016-10-03): Fix start-up race between UART & start_lua.
Fixes #1517.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [ ] The code changes are reflected in the documentation at `docs/en/*`. **N/A**

\<Description of and rational behind this PR\>
Input during startup (especially while doing initial filesystem format)
ran the risk of filling up the task queue, preventi [... cut]