nodemcu/nodemcu-firmware #1518: Somfy/TELIS driver module

vsky279, 2016-09-29:
### Missing feature
Somfy/TELIS driver module

### Justification
Possibility to control my Somfy blinds with ESP8266 using an RF transmitter (433.42 MHz) connected to GPIO pin.

The module is quite simple. It uses hw timer for precision timing (similar implementation to `gpio.serout()`).
The module: https://github.com/vsky279/nodemcu-firmware/blob/somfy/app/modules/somfy.c
Documentation: https://github.com/vsky279/nodemcu-firmware/blob/somfy/docs/en/modules/somfy.md

### Workarounds
The feature could possibly be replicated using `gpio.serout()` in its asynchronous mode but it would require to build the frame including scrumbling of data in Lua which I deem not very elegant.

### Drawbacks
I am not sure the module will find sufficient number of users. If not it does not probably worth to be merged in the main branch.

marcelstoer, 2016-10-01:
> I am not sure the module will find sufficient number of users.

You'll never know until it's actually available. I suggest you create a PR so the code can be reviewed.

vsky279, 2016-10-02:
Done. PR https://github.com/nodemcu/nodemcu-firmware/pull/1521

closed 2016-10-13 by vsky279