nodemcu/nodemcu-firmware #1520: ws2812 buffer copy and merge operations?

nwf, 2016-10-02:
It'd be nice to have direct access to memmove() on the ws2812 buffer objects rather than having to get/set in a loop in Lua.  Of course, it'd also be nice to have bitblt operations more generally, and at this point we're probably beyond something we want to be specific to ws2812 LEDs; see #1352 .

devsaurus, 2016-10-02:
> beyond something we want to be specific to ws2812 LEDs

Second that and would even see use cases beyond LEDs in general. Like marshaling data chunks for network connections in combination with `struct`.

Universal, generic operations on the buffer content with different back-end stubs for xyz LED, network, plain string?

devsaurus, 2016-11-01:
Closing this in favor of the more general discussion in #1579. Please follow up with any suggestions there.

closed 2016-11-01 by devsaurus