nodemcu/nodemcu-firmware #1524: Integrate a full coroutining framework into NodeMCU

TerryE, 2016-10-03:
### New feature
This involves the formalisation of a two lua thread framework to coding Lua applications for the ESP8266 and ESP32.
-  _**L0** the default thread_.  This will be used as currently for existing applications which don't adopt this framework.  Within the framework L0 will still continue to be used do deliver event-based tasks and execute console input.

-  _**LM** the main thread_.  This thread will persist for the life of the application and provide a procedural context that will be more familiar to Ardiuno and applications programmers
 
It would also provide some `node` API extensions to facilitate integrating this framework into library modules, and `node` library methods to simplify applications programming. 

### Justification

The ESP8266 non-OS SDK is non-preemptive, so all processing (with the exception of some time critical ISRs) executes as small atomic tasks.  The SDK still executes time-critical Wifi, DNS and LwIP activities within this task framework so tasks must complete in ~15 mSec or so to allow these tasks to complete in a timely manner.  Programmers new to the SDK and to ESP8266 have a real difficulty in switching from the sort of procedural style found with the Arduino SDK, etc. to the event-driven model used within the SDK.

The ESP32 uses a preemptive process-oriented RTOS with processing split over two processors. This means that housekeeping and drivers can and do run asynchronously. Notwithstanding this, the Lua VM based on eLua is single threaded and non-reentrant so at an applications level all Lua execution executes within a single process very much as the non-OS SDK operates.

So my design goals in suggesting this framework design are:

-  To enable existing ESP8266 event-driven Lua applications to continue to work with no or little modification  
-  To provide a unified application Lua programming across the two very different execution harnesses, and thus largely mitigate the migration issues when moving across the two ESP platforms.
-  To provide a simple to understand procedural execution model for programmers used to this programming style
-  To simplify the task of writing extension libraries for the two platforms and again unify these as much as possible. 

### Workarounds

The low level implementation could be largely implemented as is using the standard Lua corouting support.  However this implementation would have to be different for the ESP8266 non-OS and ESP32 RTOS variants. By moving the main variant handling into the platform / node layers, we can largely unify the interface at an Lua application level, and hide some of the complexities from the applications programmer.

### Discussion

Whichever execution framework is adopted, Lua itself is single threaded with asymmetric cooperative multi-tasking.  The need to have timely event/action routines means that all Lua processing still needs to be split into small atomic routines.  The best way to avoid this in Lua is to utilise its `coroutine` library, and with a little finesse we can do this in a way that is both simple to use _and_ runtime efficient.

The Lua RTS will provide a C API interface to allow any event driven routine to queue a task for execution within the Lua process:
- on the ESP8266, the SDK scheduling mechanisms effectively implement this anyway so in general Lua can be directly called from C tasks, though this has been extended by providing a node.task interface.
- on the ESP32, this interface will be need to be thread aware and reentrant using RTOS mutex / scheduling to queue tasks for execution in the Lua process.

All interactive Lua commands and any task requests are executed in the L0 thread and run uninterrupted to completion.
-  the **`init.lua`** task or subsequent event can execute a `node.mainProcess(func)` call which will create the LM coroutine thread to execute any procedural control process.  This process can then use the following `node`methods:
-  **`node.waitEvent()`** yields to the main thread, and resumes control on completion processing.  The return status is the arguments returned by the L0 resume.
-  **`node.pollEvent()`** returns immediate control (no yield is executed) with a `nil` status if no L0 tasks are queued, otherwise the status return is as per `waitEvent`.
-  **`node.allowEvent()`** yields to the main thread if any L0 tasks are queued otherwise returns immediate control.  The return status is the number of thread responses queued.  Note that in the case where L0 event tasks return a resume status, then the node API queues these so no event returns are lost.

The reason for the three variants is that the LM application code can insert `allowEvent` calls within long processing loops to facilitate adequate responsiveness for Lua L0 event-based routines,  the LM code can decouple the process of even returns to simply its structure.  The wait and poll variants allow blocking and non-blocking approaches.

It is moot whether we also add some form of "nice" algo to yield to a time delay if the LM process has run continuously for more than a specified period.

There is a default LM process which simply executes a `while (true) do node.waitEvent() end` loop.

The implementation requires a Lua 5.2 API feature `lua_yieldk()`, but back-porting this to nodemcu 5.1.5 is reasonably straight forward.

TerryE, 2016-10-05:
@pjsg, @jmattsson,....  I am taking a couple of weeks break over in Greece to close up our cottage for winter. Jan is staying home to keep an eye on the build (her turn), so I'll have lots of time to program.   What do you guys think of this idea? I don't want to make a start if there is  no support for this.

jmattsson, 2016-10-06:
I'm feeling a bit too far removed from the "regular user" base to have a strong opinion whether this is needed or not. Making the initial leap from procedural to event driven is certainly a bit of a mind bender, but it's been... a while.. since i did that. Those of you who hang out more on the forums and field user questions probably have a far better idea than I on this.

On a technical level however, don't do this because you think something different is needed for the ESP32 (or RTOS on 8266). Your task framework has slotted nicely into the RTOS (*really nicely* into the IDF actually, thanks the new system event flexibility). The underlying concept is the same whether you've got an ISR running under non-OS SDK or an RTOS task/thread - if you want to get stuff into Lua land, you post.

I'm also a bit hesitant to suggest kicking off something this large at this particular time; We already have some big things in flight that I'd like to see finished off. The new `net` module is *almost* done and has been for some time now and I'd love to see that completed, as that will help close the gap between 8266 and the 32. Same with the mbedTLS introduction that's slated for introduction when the `net` stuff is in. It would be great if you could use some of your time to help shepherd some of these things across the finish line, and lend some expertise & time to those who've been working on it.

I don't know whether any of you have been able to get your hands on any ESP32 boards (I saw Adafruit sold out overnight again), but I would also appreciate a hand with lifting modules across from `dev` onto `dev-esp32` to get it to a somewhat useful state to encourage others to contribute there.

And of course, as you mentioned over on the actual ESP32 status issue, we need to think through and decide on what we actually want to do with support for the two chips. If you could spend some of your time familiarising yourself with the ESP32, the IDF and the `dev-esp32` branch in general, that'd be great. The ESP32 Reference Manual on the Espressif site isn't complete, but there's a lot of info in there already.

TerryE, 2016-10-06:
@jmattsson Thanks. If it's useful to get your perspective. You make very valid points about leaving tasks part done and avoiding opening new complex ones to proliferate effort. I'll put this on the back burner for now.  Like you, I am entirely comfortable with the event driven paradigm.  Maybe we work up to this in two steps: first, I use the pure  Luanda approach on a serious application, and then document it and see if it helps newbies before  we implement a wrapper.

I will catch up on  net,  LWiP and the ESP32  for now.

pjsg, 2016-10-06:
@terrye If you are going to have a bunch of time free, then I think that the biggest missing piece is a supported OTA update procedure/implementation. There seems to be some code floating around, but nothing integrated....

jmattsson, 2016-10-06:
@pjsg I'd like to hold off a little while yet on that to see what happens on that front on the ESP32. I suspect we might actually want to use the Espressif way there, and we could save us some hassle if we pick the right implementation at that point :)  My $.02 on that front.

TerryE, 2016-10-06:
OK. Time to cry into my beer. I have just driven to my sister-in-law's to overnight before flying out at dawn tomorrow. I have just realised that even though I've packed my laptop,  and  two ESP8266 dev kits,  I had a total mind fart and forgot my power brick, and there's no point in my lugging a 17" laptop to Greece and back  for  3 hrs battery life  😢 so Jan is taking it home again.  No laptop; no programming.

TerryE, 2018-08-29:
This one has been sitting untouched for two years.  Given the difficulty of getting something like LFS adopted, I think that there is zero benefit case for doing this.

closed 2018-08-29 by TerryE