nodemcu/nodemcu-firmware #1531: ESP8266 12E crashes everytime on interrupt.

axkodela, 2016-10-07:
Hi everyone

I am hoping that it is not some silly mistake on my part. I coded a gpio pin as interrupt and pulled it up. Sometimes without any further action and sometimes with grounding the stated pin, the esp resets. I will enclose my schematic here. I tried connecting the pin to ground with a resistor and tried triggering on up than the down. Tried leaving it floating. I tried it on two different boards, both are esp8266 12E, one on the nodemcu devboard and another on the bare one, hand soldered. I got the nodemcu from the cloud builder. Tried several versions. Annoying part is that it works if I use esp8266-core for arduino and use c++ to attachInterrupt. Sadly my application is almost completely done for a wearable, including the screen, webserver, rtc time etc. The last part of showing a clock on the button, which is where the interrupt comes, is killing me! Hope I don't have to rewrite the whole thing in C++.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
call intcb upon pin12 being grounded.
### Actual behavior
wdt reset
### Test code

```Lua

ipin=12
last=tmr.now()

function intcb(level)
     if tmr.now()>last+100 then 
     print("hi there")
     end
     return
end

gpio.mode(ipin,gpio.INT,gpio.PULLUP)
gpio.trig(ipin,"down",intcb)

```
### NodeMCU version
Master branch built on cloudbuilder.

### Hardware
ESP8266 12E

8<------------------------ END BUG REPORT -------------------------------------

Besides the complaint, many thanks for making an awesome tool for people like me to play with.
Thanks in advance.
[image]

jmattsson, 2016-10-07:
You might be using the wrong pin here. If you want the raw ESP GPIO12 line rather than D12 on a NodeMCU board, then you should be using index 6, see: https://nodemcu.readthedocs.io/en/dev/en/modules/gpio/

D12 maps to GPIO10, which may be in use by the SPI flash if you're configured for QIO mode, which could explain the WDT reset. I tested your example on D1 / GPIO5 on my NodeMCU 0.9 devboard and it functioned as expected.

sxkod, 2016-10-07:
Hi there

Sadly,  I was using physical pin 6. I have a pin mapping image and followed
it. I tried a few other pounds including sd1, sd2 etc on the dev board. I
will run this with different pins again and post the actual reset messages.

Thanks for your quick reply.

sxkodela, 2016-10-08:
I played with it again today. My test_int.lua contains the code at the end. My init.lua is empty and I only ran the test_int.lua. It crashes even when the code was run and no input / d6 [gpio 12] was even touched. It is a nodemcu dev board. I have a 2amp, 12v power supply feeding the breadboard power supply module, that is feeding 5v to vin. With or without the external power it is the same outcome.

Any help greatly appreciated.
Thanks

`dofile("test_int.lua")

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 24300, room 16 
tail 12
chksum 0x62
ho 0 tail 12 room 4
load 0x3ffe8000, len 2256, room 12 
tail 4
chksum 0xb2
load 0x3ffe88d0, len 8, room 4 
tail 4
chksum 0xee
csum 0xee
����n��r��n|�llll`��r�l�lL�

NodeMCU custom build by frightanic.com
	branch: master
	commit: 8e48483c825dea9c12b37a4db3d034fccbcba0bf
	SSL: false
	modules: file,gpio,i2c,net,node,rtctime,sntp,tmr,u8g,wifi
 build 	built on: 2016-09-25 20:15
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
> 
`

The code for the lua file is below.
`ipin=12
last=tmr.now()

function intcb(level)
     tmr.delay(0)
     if tmr.now()>last+2000 then 
     print("hi there")
     end
     return
end

function pcb() print("from timer") end

gpio.mode(ipin,gpio.INT,gpio.PULLUP)
gpio.trig(ipin,"both",intcb)
`

devsaurus, 2016-10-08:
>  ipin=12

This code resets my module as well. As @jmattsson said, pin 12 addresses GPIO10 which interferes most likely with the onboard flash chip.
Please change to `ipin = 6`, this works for me.

sxkod, 2016-10-08:
Thanks @Mattsson and @Lauger. Now it works on D6 if I use the ipin as 6. So
the mapping I have been using is totally off :-(

https://acrobotic.com/media/wysiwyg/products/esp8266_devkit_horizontal-01.png

Thanks a lot for saving me from ending up rewriting the whole thing in C++!

closed 2016-10-09 by devsaurus