nodemcu/nodemcu-firmware #1536: Task queue overflow

pjsg, 2016-10-10:
### Expected behavior

When gpio.serout completes, the callback should run.

### Actual behavior

It sometimes doesn't.

### Analysis

The callback doesn't run when there is a lot of serial input. I suspect that the task queue (low) fills up and the post of the serout complete is dropped. This seems like another manifestation of the 'input during fs format kills the system' problem.

Can we make the serial input handler just post a single message saying that there is input and have the flag then cleared by the task level (and then checked by interrupt level). In this way, there would be a maximum of 1 message for serial input in the queue.

I realize that I could choose a different queue, but this behavior may well break other callbacks as well.

pjsg, 2016-10-11:
I *suspect* that in app/driver/uart.c the got_input only needs to be set to true if a character is received when the input buffer queue is empty. If the queue is not empty, then we must have sent a post message in the past and it is still being handled.

eku, 2016-10-11:
@pjsg setting/resetting got_input may lead to race conditions.

pjsg, 2016-10-11:
The got_input flag is local to the interrupt handler, so there are unllikely to be race conditions there. However, there may be a race condition while the base level task is consuming the last byte of input. I don't know whether this could be solved by posting the task when there are 0 or 1 bytes of pending input.

devsaurus, 2016-10-11:
A kind of handshaking should solve the issue:
The ISR contains a counter while the task maintains a copy of it, both start with the same value. When the ISR decides to post the task, it increments the counter (with wrap-around) and provides the new value as parameter to `task_post_low()`. The task updates its copy from the parameter and begins to process the input.
The decision whether to post the task or not depends on the current value of the counter and the task's copy:
- Not equal: The task has not yet started to process the most recent post. There's no point in posting it again.
- Equal: The task completed the most recent post or is currently processing that. A new post is required to ensure that the task picks up the latest buffer updates. Posting is redundant in case the task won the race and will process the new data right now, but that's ok.

I assume that the counter can be reduced to a binary toggle flag.

PR #1540 (2016-10-18): Avoid task queue overrun for serial input
Fixes #1536.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

This PR mitigates overrunning the task queue with superfluous posts for the serial input handler. Details are described in https://github.com/nodemcu/nodemcu-firmware/issues/1536#issuecomment-253062971.

#### Test case
I used the code from http://www.esp [... cut]
closed 2016-10-18 by pjsg