nodemcu/nodemcu-firmware #1544: gpio.trig() not same performance as doc

trailsandtribulations, 2016-10-20:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
gpio.trig( 8, 'both' ) would remove trigger for 'both' if exists 

### Actual behavior
gpio.trig( 8, 'both' ) raises error

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
gpio.mode( 8, gpio.INT )
gpio.trig( 8 )          -- this works
gpio.trig( 8, 'both' )  -- this bombs
-- ERROR: stdin:1: bad argument #3 to 'trig' (invalid callback type)s
```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

marcelstoer, 2016-10-20:
Our GitHub issues template asks you to provide the NodeMCU version you use. Which one is it?

Does it work if you leave out the first `gpio.trig` call?

```lua
gpio.mode( 8, gpio.INT )
gpio.trig( 8, 'both' )  -- what happens?
```

pjsg, 2016-10-21:
The documentation seems to indicate that 

```
gpio.trig(8, 'both')
```

should enable both interrupts on pin 8 using the previously specified callback, The error indicates that there is no previously specified callback.

If you want to clear the interrupt, then you do

```
gpio.trig(8)
```

I think that the behaviour is 'works as designed'

marcelstoer, 2016-10-21:
Philip's right, sorry I missed it.

> `type`: If the type is "none" or omitted then the callback function is removed and the interrupt is disabled.

> `callback`: The previous callback function will be used if the function is omitted.

closed 2016-10-21 by marcelstoer

trailsandtribulations, 2016-10-25:
> =node.info()
1 5 4 2358765 1458415 4096 2 40000000

interesting:
> gpio.mode( 8, gpio.INT )
> gpio.trig( 8, 'both' )  => OK
> gpio.trig( 8 ) => OK
> gpio.trig( 8, 'both' ) => ERROR bad arg #3