nodemcu/nodemcu-firmware #1545: HTTP request failed when trying to make a GET

bistory, 2016-10-20:
### Expected behavior
GET requests from HTTP module worked.

### Actual behavior
Before v1.5.4, GET requests were working as expected. I downloaded a new firmware and HTTP is not working anymore.

### Test code

```Lua
-- Your Wifi connection data
local SSID = "carotom"
local SSID_PASSWORD = "caranelle"

-- If connection is successful, read DHT and post
wifi.eventmon.register(wifi.eventmon.STA_GOT_IP, function(T)
    print("Config done, IP is " .. T.IP)

    http.get(string.format("http://www.google.com"), nil, function(code, data)
        if (code < 0) then
            print("HTTP request failed")
        else
            print(code, data)
        end
    end)
end)
```
### NodeMCU version
1.5.4.1-master_20161001 -> Not working
1.5.1-master_20160603 -> Working

### Hardware
Tested with ESP12F on a dev board and on the NodeMCU v2

marcelstoer, 2016-10-20:
Tracked with #1546.

closed 2016-10-20 by marcelstoer

bistory, 2016-10-20:
This bug occurs also with ThinkSpeak responses which are tiny (only the id of the insert) :)

marcelstoer, 2016-10-20:
Please add that feedback to #1546 instead.