nodemcu/nodemcu-firmware #1546: HTTP request with "large" response body fails or resets device

marcelstoer, 2016-10-20:
This is a follow-up to #1485 and a more-to-the-point description of #1545. Seems like a serious regression bug in the HTTP module or some SDK characteristic the HTTP module relies upon.

### Expected behavior
HTTP GET to an HTTP resource should return the HTTP resource.

### Actual behavior
Depending on the response size (I believe) this either works, fails or resets the device.

### Test code
```Lua
http.get("http://nodemcu-build.com", nil, function(code, data)
  if (code < 0) then
    print("HTTP request failed")
  else
    print(code, data)
  end
end)
```
- http://frightanic.com/jquery-highlight.js ✅ , 2.7 KB
- http://frightanic.com 🔴 , 17,7 KB
- http://nodemcu-build.com 🔴 ♻️ , 6.6 KB
- http://nodemcu-build.com/about.php 🔴 ♻️ , 2.8 KB
- http://nodemcu-build.com/jumbotron-narrow.css ✅ , 433 B
- http://www.google.com, 🔴 , 66.9 KB
- http://clients5.google.com/pagead/drt/dn/, ✅ , 270 B

🔴 : fails
♻️ : reset

However, and that's the big mystery for me http://twitter.com/nodemcu with >50kB response size does NOT fail. Since the response is multiple packets printing `string.len(data)` never prints the correct value, though (it's an arbitrary value).

### NodeMCU version
Current `master` and `dev` with SDK 1.5.4.1. According to #1545 this worked fine with 1.5.1. Did the SDK used to buffer responses in earlier versions?

### Hardware
NodeMCU devkit 1.0.

bistory, 2016-10-20:
I have to add that this problem also occurs on thingSpeak which returns tiny responses but needs long URI's (I haven't tested yet if this problem is related to URI)

devsaurus, 2016-10-20:
A quick browse through `app/http` shows that there is an upper limit for buffering. Does it apply to your cases?
You could enable `HTTPCLIENT_DEBUG_ON` and look for major issues from the client code.

marcelstoer, 2016-10-20:
> You could enable `HTTPCLIENT_DEBUG_ON`

Hhmmm...\*scratching my head\*

I had the global debug in `user_config.h` on (`#define DEVELOP_VERSION`) and was seeing some debug messages. However, I never bothered to check the `httpclient.c` code. After I did I realized how many more debug messages I _should_ have seen. 
`httpclient.h:13` says `#if defined(GLOBAL_DEBUG_ON)` but a reverse grep through the entire code base didn't find `GLOBAL_DEBUG_ON` anywhere else. A bug?

luismfonseca, 2016-10-20:
`0x2000` hexadecimal, so it's `8192` in decimal.

It would explain why files bigger than 8KB fail, but since it also fails when they are smaller, it's probably irrelevant.

The list you have here doesn't have any `https`. Does this mean that this also happens regardless of https?

marcelstoer, 2016-10-20:
> `0x2000` hexadecimal, so it's `8192` in decimal.

That much I know 😉 I guess that's bytes then...

> The list you have here doesn't have any `https`. Does this mean that this also happens regardless of https?

That's where #1485 started but it turned out it's got nothing to do with secure vs. insecure connections.

devsaurus, 2016-10-20:
Enabled `DEVELOP_VERSION` in `user_config.h` and `HTTPCLIENT_DEBUG_ON` in `httpclient.h`and exec Marcel's test case against different hosts yields different failure modes:

### nodemcu-build.com

```
hostname=nodemcu-build.com
port=80
method=GET
path=/
DNS request
DNS pending
> DNS found nodemcu-build.com 217.26.52.40
Connected
Sending request header
All sent
Disconnected
hostname=nodemcu-build.com
port=443
method=GET
path=/
DNS request
DNS pending
DNS found nodemcu-build.com 217.26.52.40
client handshake start.
client handshake ok!
Connected
Sending request header
All sent
client's data invalid protocol
Error: invalid protocol message
Disconnected with error
Disconnected
Chunk Size:8192
Chunk Size:0

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

### frightanic.com
```
hostname=frightanic.com
port=80
method=GET
path=/
DNS request
DNS pending
> DNS found frightanic.com 217.26.52.40
Connected
Sending request header
All sent
Response too long (8713)
Disconnected
http_status=-1
HTTP request failed
```

devsaurus, 2016-10-20:
> It would explain why files bigger than 8KB fail, but since it also fails when they are smaller,

Looking at the messages for nodemcu-build.com, I'd guess that the initial GET with small response size would pass ok, but it gets redirected to port 443 and fails then -> SSL specific?

marcelstoer, 2016-10-20:
> Enabled `DEVELOP_VERSION` in `user_config.h` and `HTTPCLIENT_DEBUG_ON` in `httpclient.h`

I have some questions about this (my first comment here) but I'd like to discuss them separately -> #1548.

>  for nodemcu-build.com, I'd guess that the initial GET with small response size would pass ok, but it gets redirected to port 443 and fails then -> SSL specific?

Earlier I thought that it may have to do with SNI or the lack of support thereof (#1484 and #1090). However, http://nodemcu-build.com/jumbotron-narrow.css works just fine even though it also gets redirected.

I realized that there are a number of `HTTPCLIENT_DEBUG` messages (not just response too long) which shouldn't be _debug_ messages but regular error messages i.e. printed unconditionally.

bistory, 2016-10-21:
Hi,

Some precisions about my previous post : here is a snippet which feeds a dummy channel on ThingSpeak (https). The reply is tiny but it fails anyway...

```Lua
-- Your Wifi connection data
local SSID = "TEST"
local SSID_PASSWORD = "PASSWORD"

-- If connection is successful, read DHT and post
wifi.eventmon.register(wifi.eventmon.STA_GOT_IP, function(T)
    print("Config done, IP is " .. T.IP)

    http.get(string.format("https://api.thingspeak.com/update?api_key=%s&field1=%d.%03d&field2=%d.%03d&field3=%.4f",
              "Z30NN8EZ6YWTAORK",
              19,
              6,
              43,
              6,
              3.8078), nil, function(code, data)
        if (code < 0) then
            print("HTTP request failed")
        else
            print(code, data)
        end
    end)
end)
```

marcelstoer, 2016-10-21:
@bistory I bet yours is an SSL/TLS issue (and not the same as #1545). I don't want make it sound as if it wasn't important but I don't want loose the focus on this issue, happened before. You can use the ThingSpeak API w/o HTTPS, right?

bistory, 2016-10-21:
No I can't... Is there already a bug opened about SSL/TLS or should I open a new bug ?

marcelstoer, 2016-10-21:
http://api.thingspeak.com/update?api_key=Z30NN8EZ6YWTAORK&field1=19.6&field2=43.6&field3=3.8078 works fine. It probably isn't even a bug but an SDK limitation, see http://nodemcu.readthedocs.io/en/latest/en/modules/http/ -> SSL/TLS support.

bistory, 2016-10-21:
That doesn't explain why the exact same code worked before v1.5.4 and doesn't work anymore... Or the SDK limitations changed ?

pjsg, 2016-10-24:
I have a fix to part of the problem -- the resetting part. It now doesn't reset when talking to nodemcu-build.com.    However, it still doesn't work! I also got fed up and added a version of printf that would work without using large amounts of stack, and could print out arbitrary strings. Doesn't support all the flags fully, but works for debugging.

PR #1558 (2016-10-27): Fix for httpclient resetting the system (maybe)
Partially Fixes #1546.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ ] I have thoroughly tested my contribution.

The issue was that the reconcb called the disconnect cb. This was bad as it freed up all the control blocks and didn't disconnect the connection. I switched it over to called the timeout callback which does try and disconnect the connecti [... cut]
PR #1568 (2016-10-29): Change HTTP failures from debug to error messages
Addendum to #1558, fixes #1546

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
closed 2016-10-29 by marcelstoer