nodemcu/nodemcu-firmware #1548: How is our debug mode supposed to work?

marcelstoer, 2016-10-20:
In my understanding our debug mode works as follows (what I _think_ I remember):

- If you enable `#define DEVELOP_VERSION` in `user_config.h` _all_ debug messages are turned on.
- The there's a module specific switch e.g. `HTTPCLIENT_DEBUG_ON` in `httpclient.h` which only turns on the debug messages for a particular module. The global switch should enable the module specific switches one way or the other.

However, it isn't implemented correctly/consistently or my understanding is wrong.

Why would I need both `DEVELOP_VERSION` in `user_config.h` and `HTTPCLIENT_DEBUG_ON` in `httpclient.h` to see HTTP debug messages?
Why does `httpclient.h` check for `#if defined(GLOBAL_DEBUG_ON)` but a reverse grep through the entire code base didn't find `GLOBAL_DEBUG_ON` anywhere else?

pjsg, 2016-10-21:
It is all a bit of a mess. I find that I have to edit the code to enable debug in a module....

djphoenix, 2016-10-21:
I think best way for debugging enable/disable is as proposed:
- One global switch that enables just all
- Per-module switches

Also maybe it will be nice if we're support multi-level debugging (as in LWIP), so config will look like that:
```c
#define DEBUG_GLOBAL DEBUG_WARN
// Overrides
#define DEBUG_NET    DEBUG_OFF
#define DEBUG_MQTT   DEBUG_INFO
// ...
```

marcelstoer, 2016-10-23:
Thanks Philip for the confirmation. It's good to hear that from someone who's actually qualified to judge our source code.

How to proceed, in general and with this issue? Shall we use this issue to discuss and define how we _actually_ want it work? I guess even someone like myself (i.e. not intimately with C) could then go through the code and fix it one by one.

jmattsson, 2016-11-14:
I never use DEVELOP_VERSION; whenever I need debug prints in a module it's something that gets enabled in that module only. Global debug message are of no interest to me, they just drown out the bits I'm interested in. Just my $.02

marcelstoer, 2016-11-14:
Johny, I agree but I still wouldn't want to drop it. Would you agree that what we recently fixed for the HTTP module (`httpclient.h`) would be a sensible pattern for all modules:

```c
#if defined(GLOBAL_DEBUG_ON) // should actually be DEVELOP_VERSION
  #define HTTPCLIENT_DEBUG_ON
#endif
#if defined(HTTPCLIENT_DEBUG_ON)
  #define HTTPCLIENT_DEBUG(format, ...) dbg_printf("HTTP client: "format"\n", ##__VA_ARGS__)
#else
  #define HTTPCLIENT_DEBUG(format, ...)
#endif
#if defined(NODE_ERROR)
  #define HTTPCLIENT_ERR(format, ...) NODE_ERR("HTTP client: "format"\n", ##__VA_ARGS__)
#else
  #define HTTPCLIENT_ERR(...)
#endif
```

jmattsson, 2016-11-14:
Does anyone actually find a "enable all debugging across the entire firmware" option to be useful? Honest question. Personally I don't, but my views aren't necessarily shared all that wide.

There's also a whole bunch of unnecessary string duplication which would bloat the firmare the way those macros are written; Would it not be better to
```
static const char log_prefix[] = "HTTP client: ";
...
#define HTTP_CLIENT_ERR(format, ...) NODE_ERR("%s"format"\n",  log_prefix, ##__VA_ARGS__);
```
so only the `%s` and `\n` are added as "overhead" to each log format string?

marcelstoer, 2016-11-15:
> Does anyone actually find a "enable all debugging across the entire firmware" option to be useful? 

Probably not for firmware developers but for NodeMCU users (me speaking from nodemcu-build.com perspective). If people are stuck with an issue it may help us to get the fully picture.

> so only the %s and \n are added as "overhead" to each log format string?

True, that'd definitely be an improvement.

TerryE, 2016-11-18:
> I never use DEVELOP_VERSION; whenever I need debug prints in a module it's something that gets enabled in that module only. Global debug message are of no interest to me, they just drown out the bits I'm interested in.

Snap.

jmattsson, 2016-11-18:
> Snap.

Do I read that as "that's the dumbest thing I've read today" or "yeah, I totally concur"? :)

TerryE, 2016-11-18:
Sorry, I totally concur.

C.f. https://youtu.be/hxaRJt-x9sM 😁

marcelstoer, 2016-11-19:
I understand and agree `DEVELOP_VERSION` has little to no value for firmware developers. Does it bother you enough to call for its removal for the rest of the community or can you live with it?

If so I'll close this issue as we seem to have reached a kind of consensus. All ideas went into https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/http/httpclient.h#L13-L27 and I'll document that pattern somewhere.

closed 2016-12-31 by marcelstoer