nodemcu/nodemcu-firmware #1551: Add support to capture the time that a GPIO trigger is generated

pjsg, 2016-10-22:
### Justification
It will make decoding some protocols easy.

### Workarounds
You can try and get the time when the callback is scheduled, but there are unpredictable delays.

pjsg, 2016-10-22:
I found the PR - #1017. This does something rather different and makes use of a clock. I wonder if an API that (say) recorded the times of up to <n> trigger conditions on a pin, and then returned all the data in one go. Maybe with both a limit on the number of edges reported and a timeout if nothing happens for a while.

i.e.

```
gpio.serin(pin, 'both', 100, 100 * 1000, function (result, initialedge)
  -- result is a list of timestamps of up to 100 'both' edges. Inactive timeout of 100 * 1000 microseconds.
  -- initialedge indicates if first edge was 'up' or 'down'
end)
```

This would start monitoring that pin and wait (potentially forever) until it changed state. Then there is a 100ms max timeout of inactivity. You could call `gpio.serin(pin)` to cancel the operation.

This would probably work down to pulses of tens of microseconds.

Does this seem reasonable?

dvv, 2016-10-24:
I'd try that.

closed 2019-08-04 by stale[bot]