nodemcu/nodemcu-firmware #1552: NodeMCU social media process

vowstar, 2016-10-22:
Now  https://facebook.com/NodeMCU is under our control, the page  https://facebook.com/NodeMCU is deleted and I've change https://www.facebook.com/NodeMCU-618537718314559 to  https://www.facebook.com/NodeMCU.

Dear Collaborators, please send me your facebook link to me <vowstar@nodemcu.com> so I can invited you as administrator, thanks! 

@nodemcu @marcelstoer @devsaurus @pjsg @TerryE @dnc40085 @funshine 

* But NOTE: Twitter(https://facebook.com/NodeMCU) not under our control, we can send mail to twitter together to report this. *

pjsg, 2016-10-22:
I think you mean https://twitter.com/NodeMCU

vowstar, 2016-10-22:
Thanks @pjsg, I've fixed.

marcelstoer, 2016-10-22:
@vowstar Thank you so much for all your efforts! 

If you go to https://twitter.com/NodeMcu there's a 'Report NodeMcu' link in the context menu that pops up if you click the little sprocket wheel icon on the right side in the header. Besides that account being from an imposter you could as well report a potential copyright infringement on the logo from https://github.com/nodemcu/nodemcu-devkit/blob/master/Images/NodeMCU-Style5.jpg which is under MIT license and probably shouldn't have been altered the way it was. It may make an impact if we lots of people report that account.

vowstar, 2016-10-22:
@marcelstoer I've tried to report, but the https://twitter.com/NodeMcu changed all context, and no relationship with nodemcu. It have a long road to build nodemcu social media.

GerwinJanssen, 2016-10-22:
@vowstar If you're girlfriend did not attack me on WeChat some days ago. This was never happend. This was my reaction to her actions. She is all talking longer bad about me... Maybe is that reason other people don't like her... I also try to contact you. Two days ago but you did not respons to me. I had a solution for you. You also know that I was talking with Zeroday about making a Organizations page of this github. You need to thank her for all these problems!

vowstar, 2016-10-22:
@GerwinJanssen 
I used to treat you as a trusted friend, but what you do makes me feel very disappointed. 

* I can't believe you changed website:

[image]

* I can't believe you post these content:

[image]

I'm very surprised by what happend in these days.

My wife has been supportive of me, and I also stand by my wife's side. If She and I did something wrong, you can hate us, but nothing wrong with NodeMCU open source community.

vowstar, 2016-10-22:
@GerwinJanssen have called to me and we talkabout many, and @GerwinJanssen will contant @marcelstoer and give twitter access to him, I think this issue can closed.

closed 2016-10-22 by vowstar