nodemcu/nodemcu-firmware #1553: mqtt keep alive timer is not correct

lepass7, 2016-10-22:
### Expected behavior
When the esp8266 loses wifi communication it should activate the m:on("offline" ... after 120 Seconds (according to the following example)
### Actual behavior
The offline event activates randomly and after 10-15 minutes
### Test code
```Lua
-- init mqtt client with keepalive timer 120sec
m = mqtt.Client("clientid", 120, "user", "password")

-- setup Last Will and Testament (optional)
-- Broker will publish a message with qos = 0, retain = 0, data = "offline" 
-- to topic "/lwt" if client don't send keepalive packet
m:lwt("/lwt", "offline", 0, 0)

m:on("connect", function(client) print ("connected") end)
m:on("offline", function(client) print ("offline") end)
```
### NodeMCU version
master
powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

### Hardware
esp8266-12e

NicolSpies, 2016-10-22:
I experience the same and I link this to the method used in #1555 to determine if a MQTT connection is present before trying to publish a message.

pjsg, 2016-10-23:
From looking at the code, it appears that the timeout handling is missing. After the keepalive time it sends the PINGREQ -- which should the cause the server to send PINGRESP. The client is *supposed* to wait for up to keepalive time for the PINGRESP and abort the connection if nothing is received. However, I can't see the code that does this...

pjsg, 2016-10-23:
If somebody can try this fix #1557 in their test environment, that would be great. It appeared to work with a simple test case....

NicolSpies, 2016-10-23:
@pjsg, I have 3 units in the field connected via 3G that experience MQTT disconnects due to the connection. I will test on the units. Unfortunately can only get to them in a few days to upload new build.
Agh!!, I forgot, you now my challenge, I will only be able to test after a merge to dev enabling me to do a custom build.

marcelstoer, 2016-10-25:
> Agh!!, I forgot, you now my challenge, I will only be able to test after a merge to dev enabling me to do a custom build.

Sounds vaguely familiar...For some reason the Docker image doesn't work for you? I'll send you an email with firmware of the PR.

closed 2016-11-08 by marcelstoer

lepass7, 2016-11-15:
@pjsg I tried fix #1557 but unfortunately the keep alive timer is still not working correct.

PR #1557 (2016-11-08): Make the MQTT PING functionality work better.
Fixes #1553.
- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

When the keepalive PING is sent, the client is supposed to wait for up to the keepalive timer for a PINGRESP. If it doesn't get it, then the connection should be aborted. This now happens.

It should be noted that the timer passed in to the Client con [... cut]