nodemcu/nodemcu-firmware #1554: feature request: tmr.fetch() => tmr no, tmr obj - get free tmr no / obj

trailsandtribulations, 2016-10-22:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

### Actual behavior

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
-- add code here
```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature
A uniform way for code to grab a timer without one function stepping on other code's toes
### Justification
Having a core function to pull an available tmr will encourage code to get timers in a uniform and safe manner.
### Workarounds
In MY code I use a `for i = 0,6 do if not tmr.state(i) then ...` to grab a timer. BUT what assurances do I have that someone else's code is being equally polite. 
8<------------------------ END FEATURE REQUEST --------------------------------

devsaurus, 2016-10-22:
You want to use [dynamic timer objects](http://nodemcu.readthedocs.io/en/master/en/modules/tmr/#tmrcreate).

closed 2016-10-22 by devsaurus

trailsandtribulations, 2016-10-25:
right