nodemcu/nodemcu-firmware #1556: WPS support

ksrm, 2016-10-22:
Is there any plan to add support for WPS (auth via button)? This is crucial for applications where you have no other method of getting an SSID/password onto a device, so I am a little bit surprised that there is no support for it. The ESP8266 Arduino core (https://github.com/esp8266/Arduino) has support for it.

If nobody else is working on this I will make a start, but I'm interested to know if there is some reason that this hasn't been added yet.

karrots, 2016-10-22:
I can't speak for official reasons but [end user setup](https://nodemcu.readthedocs.io/en/master/en/modules/enduser-setup/) is one way to get an SSID programmed into a device without using WPS.

Additionally, WPS is [severely broken](https://www.us-cert.gov/ncas/alerts/TA12-006A) from a security perspective.

marcelstoer, 2016-10-23:
Even if we wanted to support WPS, there are good reasons not to (https://github.com/nodemcu/nodemcu-firmware/issues/778#issuecomment-158203376), I feel the support should be in the Espressif SDK.

PR #1694 (2017-01-09): WPS functionality added
Fixes #1556 .

- [X] This PR is for the `dev` branch rather than for `master`.
- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [X] I have thoroughly tested my contribution.
- [X] The code changes are reflected in the documentation at `docs/en/*`.

Although there seems to be some opposition against using WPS, I still think it is worthwhile making it available to whomever wants to use it.
Please bare in mind that my software/Github skills are  [... cut]
closed 2016-10-23 by marcelstoer