nodemcu/nodemcu-firmware #1559: one UDP and one TCP servers

vadimostanin, 2016-10-24:
Is it possible to have one udp and one tcp server?
I want to make discovering of ESP by UDP, and make data transferring by TCP
```
uart.setup(0,9600,8,0,1)

print("Start ESP8266")

wifi.setmode(wifi.STATION)
wifi.sta.config("essid","password")
wifi.sta.connect()

tmr.alarm(0, 1000, 1, function()
    if wifi.sta.getip() == nil then
        print("Connecting to AP...")
    else
        print('IP: ',wifi.sta.getip())
        registerServer()
        tmr.stop(0)
    end
end)

function registerServer()
    if srv ~= nil then
      srv:close()
    end
    if srv2 ~= nil then
      srv2:close()
    end
    print("before create server")
    srv = net.createServer(net.UDP)
    srv:on("receive",
                function(socket, request)
                    print(request)                    
                    socket.close()
                end)
    srv:listen(33249)

    srv2 = net.createServer(net.TCP)
    srv2:listen(33248,function(conn)
      conn:on("receive",function(socket,request)
        print( request )
        socket.close()
      end)
      conn:on("sent",function(conn) conn:close() end)
    end)
end
```

NicolSpies, 2016-10-24:
Hi, please refer to #1080 where your question is addressed

marcelstoer, 2016-10-24:
> Is it possible to have one udp and one tcp server?

Yes. For further user support consider #1010.

closed 2016-10-24 by marcelstoer