nodemcu/nodemcu-firmware #1560: 1-wire conditional rom search (0xEC)

dvv, 2016-10-24:
8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

1-wire conditional rom search (0xEC)

### Justification

Reduce polling rate when querying devices for changes while preserving small report latency.

### Workarounds

None AFAIK.

8<------------------------ END FEATURE REQUEST --------------------------------

TerryE, 2017-03-26:
@dvv, Surely OW search is something that you do once in the power up sequence to enumerate your OW devices.

 Can you explain why this feature is preferable to the existing implementation, or even give a use case where it is needed?

dvv, 2017-03-26:
I've been working on a 1-wire 16-channel switch with state feedback based on beautiful https://github.com/orgua/OneWireHub. Conditional search would allow to not _poll aggressively_ for all switches states but to query for those only which have changes.
It's not of priority right now.

karrots, 2017-03-26:
Attaching app note for reference: https://www.maximintegrated.com/en/app-notes/index.mvp/id/187

TerryE, 2017-03-26:
@dvv Vladimir, I was pretty unhappy with the current DS18B20 implementation which Ray did in the very early days probably before he'd gained a good incite into Lua programming practice and you to write SDK-friendly modules.  I've rewritten my version  to `ow.target_search(pin, family)` to limit the enumeration space to a single family, and I make a lot more use of event and timer driven + `node.task.post` callbacks to remove `tmr.wdclr()`nasties. 

In terms of the OW library itself, it currently doesn't implement ALARM_SEARCH (0xEC) and this is the only search function that I would envisage using at anything other than power up.

One the Esp8266 -- and unlike the Arduino -- if you have any extended period activity such as a search which can take up a second for a larger number of devices, then you _must_ break your code into separate tasks using `tmr` or `node.task.post()` callbacks.  The current OW search function essentially implements the Maxin application note on how to execute searches,  

You can, of course, implement each `ow.search()` in its own Lua task as I do in my own DS18B20 module.

TerryE, 2019-04-10:
This issue isn't so much one of the OW API, but the C DS18B20 library which is now retired.

closed 2019-04-10 by TerryE