nodemcu/nodemcu-firmware #1563: Makefile reorganization

djphoenix, 2016-10-25:
Now our Makefile:
- is **so long** (over 350 lines of code that looks very common in parts)
- uses recursion (**ouch!**) to subdirs, with parent inclusion (*oops...*)
- doesn't support parallel build (because recursion)
- generates **very huge** log
- contains some magic that can be understood only by creators

So, I see the way to fix it:

1. Rewrite build rules with `@`-rules and `echo`-es (with `VERBOSE` flag that enables full log)
2. Split main Makefile to:
     - **config.mk** - versions, links, debugging options, etc
     - **local.mk** - `.gitignore`d part with example how to override some default values (defined in config)
     - **deps.mk** - dependencies download (like SDK, patches and other)
     - **common.mk** - some definitions like `MakeLibrary`/`MakeImage`
     - **rules.mk** - rules to build objects / deps from source files
     - **flash.mk** - rules to flash built firmware to ESP (adding support to autodetect serial ports and configure port/baudrate in config files)
3. Remove recursion and replace it with components based on `component.mk` in each component-related directory, with support to override some values for specific component
4. Add CPU count autodetection on common platforms, and automatic enable parallel build if supported

jmattsson, 2016-11-14:
The thing is, the Makefile is heavily based on the official Espressif project template. While it's not exactly the cleanest, the one big advantage it has (other than working), is that it's easy for other ESP8266 developers to "migrate" to the NodeMCU project.

Any full-featured Makefile will always contain some "magic" which is only readily apparent to its creator(s).

I'm not saying we shouldn't consider this, but I also think we have more important things to focus our efforts on first. My $.02, I don't know what the others think. @TerryE @devsaurus @pjsg perhaps would like to chime in?

closed 2018-02-24 by marcelstoer

TerryE, 2018-02-24:
@djphoenix Yuri, I've had to take this whole build process apart / understand it to embed luac in it.  There are some improvements that we could do to tidy up oddities, but replacing it entirely is going to be a pretty large task.  I guess this needs a FAQ overview somewhere.