nodemcu/nodemcu-firmware #1569: How to use the gpio 11 and 12?

yanzixiang, 2016-10-29:
gpio.mode(11,gpio.OUTPUT)
gpio.mode(12,gpio.OUTPUT)

cause my board to reboot,
i wonder why.
i use this build
NodeMCU custom build by frightanic.com
	branch: master
	commit: 7b83bbb2ea134cd85ac9d63108603cc02c4e20f7
	SSL: false
	modules: bit,file,gpio,net,node,tmr,uart,wifi
 build 	built on: 2016-10-29 08:37
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

oyooyo, 2016-10-29:
The GPIOs with index 11 (which is actually GPIO9) and index 12 (which is actually GPIO10) are usually used to access the flash memory in QIO mode and cannot be used. If you play around with these GPIO pins, you will actually mess around with the flash memory interface, which will cause reboots and other errors.                                               
                                                                                    
It is possible to access the flash memory using DIO mode, in which case these pins are not used and thus would be available. But that's usually a huge challenge - google "ESP8266 GPIO9 GPIO10" to find out more about this topic.

marcelstoer, 2016-10-29:
Also see #1010.

closed 2016-10-29 by marcelstoer

yanzixiang, 2016-10-30:
i do NOT use the flash memory out of the chip,is there any method to disable the SPI bus,
neither QIO nor DIO.