nodemcu/nodemcu-firmware #1577: DHCP Server "Discover->Offer" Failed with AP mode

kicer, 2016-10-31:
### Expected behavior
At master branch, SOFTAP mode can't got IP with WiFi Adapter (Mediatek Ralink MT7601) which plugged in my Win10 PC. 

### Actual behavior
The wireshark only shows the DHCP Discover every 5~10s, no DHCP Offer/DHCP Request/DHCP ACK completely. [wireshark pcapng](https://github.com/kicer/nodemcu-firmware/raw/master/ap_dhcp_error/master_dhcp_error_withFF46_20161031.pcapng)
After revert `FF46` commit, the DHCP flow pass and connect normally. But wireshark will report `ETHERNET FRAME CHECK SEQUENCE INCORRECT` with `DHCP Offer` package. [wireshark pcapng](https://github.com/kicer/nodemcu-firmware/raw/master/ap_dhcp_error/master_dhcp_error_withoutFF46_20161031.pcapng)

### Test code
```Lua
math.randomseed(tostring(tmr.now()+node.chipid()):reverse():sub(1, 6))
wifi.setmode(wifi.SOFTAP)
cfg = {
    ssid="XXX-" .. math.random(0, node.chipid()),
    pwd="12345678",
    auth=wifi.OPEN,
    channel=6,
    hidden=0,
    max=4,
    beacon=100
}
wifi.ap.config(cfg)
```
### NodeMCU version
`master Latest commit 7b83bbb  on 1 Oct`

### Hardware
ESP-12F

### Attachment

devsaurus, 2016-11-02:
Had a look at your wireshark dump but found no fundamental differences between the DHCP Discover packet from your Win10 client compared to my machine. I was even able to manually send your dumped discover payload and get a DHCP Offer reply from the ESP.

There is however one remarkable observation, and it appears no matter which DHCP Discover packet I send to the ESP: The first discover that's received after booting the ESP gets answered by the dhcp server, but there's no reply if subsequent discover packets are sent (manually). This makes me assume that the dhcp server requires some kind of cycle starting with discover, offer, etc. up to release. After that it happily replies again to discover.
I've got no idea whether this is intended or some sort of erroneous behaviour - would need to dig into the RFCs...

In the meantime please check your wireshark logs for multiple discover packets from your Win10 PC. I.e. after ESP boot, are there additional DHCP Discover messages after the first one and before the complementing DHCP Release?

kicer, 2016-11-03:
@devsaurus, thanks for your replay.

I have enabled the `DHCPS_DEBUG` and `DEVELOP_VERSION` macro and change the `dhcps_msg.options` size for debug. It maybe `(IP protocol pad size > 0) or (IP protocol header > 20)` is the reason why the DHCP Offer package dropout.

When reduce the options size to **(=1496-20-8-236)**, then get the DHCP Offer `(Ethernet II pkg length=1514, padSize=4, header=20)` replay from ESP. [log&pcapng](https://github.com/kicer/nodemcu-firmware/raw/master/ap_dhcp_error/ap_dhcp_error-MTU1496.zip)
When reduce the options size to **(=1498-20-8-236)**, then get a DHCP Offer `(just once! Ethernet II pkg length=1512, padSize=0, header=20)` replay from ESP, but not get the DHCP Ack package. [log&pcapng](https://github.com/kicer/nodemcu-firmware/raw/master/ap_dhcp_error/ap_dhcp_error-MTU1498.zip)

kicer, 2016-11-07:
Have test use the macbook, DHCP flow passed.
Wireshark dump shows the DHCP Offer/DHCP Ack package size is **(=14+1510+4)**. So the issue is caused by this 4 pad bytes. Win PCs drop the packages which CRC error or size>MTU?

[macbook pcapng](https://github.com/kicer/nodemcu-firmware/raw/master/ap_dhcp_error/dhcp_error_macMTU1500_20161107.pcapng)

ikarus23, 2017-03-02:
Bump. Same issue here. I can't get an IP address from my ESP8266 (using the latest master branch).

unrealkate, 2017-04-22:
Getting same problem. DHCP server does not work on latest firmware. Does not matter how many times you write wifi.ap.dhsp.start() and configure it. Windows sens 1-2 DHCP packets, nodemcu sends nothing back! The only way to connect is to set static ip 192.168.4.1 manually on PC. It is not an issue of network adapter because I have a firmware that i've built long ago on 2016-07-17 (before SDK 2.0) it works but lacks so many features. The firmware that does not work looks like this:
`NodeMCU custom build by frightanic.com
	branch: master
	commit: 22e1adc4b06c931797539b986c85e229e5942a5f
	SSL: false
	modules: adc,bit,file,gpio,i2c,net,node,pwm,rtcmem,sigma_delta,spi,tmr,u8g,uart,wifi
 build 	built on: 2017-04-22 20:24
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)
`
Help! It is a real pain working with nodemcu - i need to change my ip address from static to dynamic and backward everytime i connect to nodemcu and internet (and no, i need it in ap mode).

unrealkate, 2017-04-23:
Found an android phone and connected it to nodemcu, DHCP worked fine with latest firmware. But it does not work with Windows 8.1 (which works with any routers just fine, so it's not my windows). Launched wireshark: windows sends DHCP packets and gets not response from nodemcu. I think nodemcu sends something, but it filtered out and dropped by a driver or on hardware level on my Windows laptop. Corrupted packets? bad checksum or size?

mthaak, 2017-05-02:
I have the same problem. DHCP worked indeed fine with 1.5.1 but not anymore with 2.0.0. Also it seems random because on some ESP8266 devices with the same firmware it is sometimes still possible to connect to the SoftAP and get a valid IP address. 
In some cases where I do not get an IP address immediately, it helps to wait a while. If I then try again it sometimes works after all.
Has someone solved this yet?

mthaak, 2017-05-02:
For me DHCP works when I revert commit ff46abe61f2e4d380a90d63617b002967bf8ffae such that we have `uint8_t options[312]` just like @kicer found. I am not sure if this is a good solution for all cases but it worked for me.

joysfera, 2017-05-19:
I can confirm the DHCP issue. It seems to be related to certain HW configurations (happens with the SONOFF modules but ESP-07 seems to be unaffected).

devsaurus, 2017-05-19:
I can't reproduce the problem here. Could someone please check whether the changes here https://github.com/devsaurus/nodemcu-firmware/tree/dhcp_send_fix solve the issue?

devsaurus, 2017-05-21:
@mthaak, @joysfera, @RafNie any change to test drive the branch linked in my previous comment?

This bugger is still around in ESP32 IDF: espressif/esp-idf#631.

RafNie, 2017-05-22:
@devsaurus  I've tested yours change today and it seems it resolve my problem described in #1979
Each of my devices connects almost immediately. Thanks for fix.

Will it be delivered into the maintrack?

devsaurus, 2017-05-22:
Thanks a lot for testing @RafNie! I'll polish the code a bit and will post it as a PR soon.

joysfera, 2017-05-22:
I couldn't have tested it yet because I experienced the issue on specific HW only (Sonoff, which might be ESP8285?) and it happend only somewhere/sometimes. So it isn't reproducible well enough to confirm that this patch helps, sorry.

joysfera, 2017-05-23:
Damn, today I encountered it even on ESP-07. I think it got invoked by launching browser and trying to load http://192.168.4.1/ page even before the connection to WiFi.
Wish this fix was part of the version 2.1.0 that was released two days ago.

PR #1985 (2017-05-25): dhcps: ensure standard message length when sending response to clients
Fixes #1577, #1979.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have **sort of** tested my contribution, courtesy of @RafNie.

This change is supposed to address the long-standing issue with certain clients that cannot cope with the extended response size of the dhcp server. I rate this fix somewhere between a workaround  [... cut]
closed 2017-05-25 by marcelstoer