nodemcu/nodemcu-firmware #1578: enduser_setup can fail to cleanup tcp listen sockets

karlp, 2016-10-31:
### Expected behavior
enduser_setup.start() doesn't always succeed in being called again.  Calling stop() should cleanup properly, and you should be able to call start() again and have it "work"

### Actual behavior
Sometimes, after successfully connecting to the chosen AP, if you then interactively try and start again, you can leak the tcp socket, and are unable to restart the process without resetting.

### Test code
```Lua
function es_ok()
        print(string.format("Connected to wifi ssid: %s our IP: %s", wifi.sta.getconfig(), wifi.sta.getip()))
end
function es_bad(err, str)
        print("es failed, err: ", err, " str: ", str)
end
function es_dbg(str)
        print("es dbg: ", str)
end
> wifi.setmode(wifi.SOFTAP)
> enduser_setup.start(es_ok, es_bad, es_dbg)
es dbg: 	340: 	enduser_setup_check_station_start
es dbg: 	1219: 	enduser_setup_ap_start
es dbg: 	1348: 	enduser_setup_dns_start
es dbg: 	1154: 	enduser_setup_http_start
es dbg: 	1163: 	http_start bind failed <<<
es dbg: 	1526: 	enduser_setup_stop
es dbg: 	1211: 	enduser_setup_station_stop
es dbg: 	1400: 	enduser_setup_dns_stop
es dbg: 	1200: 	enduser_setup_http_stop
es dbg: 	1317: 	enduser_setup_free
> enduser_setup.stop()
> enduser_setup.stop()
> enduser_setup.stop()
> wifi.setmode(wifi.SOFTAP)
> enduser_setup.start(es_ok, es_bad, es_dbg)
es dbg: 	340: 	enduser_setup_check_station_start
es dbg: 	1219: 	enduser_setup_ap_start
es dbg: 	1348: 	enduser_setup_dns_start
es dbg: 	1154: 	enduser_setup_http_start
es dbg: 	1163: 	http_start bind failed   <<<
es dbg: 	1526: 	enduser_setup_stop
es dbg: 	1211: 	enduser_setup_station_stop
es dbg: 	1400: 	enduser_setup_dns_stop
es dbg: 	1200: 	enduser_setup_http_stop
es dbg: 	1317: 	enduser_setup_free
> enduser_setup.stop()
> 
```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.
1.5.4.1-master_20161001  + local changes to enable ENDUSER_SETUP_DEBUG_ENABLE

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.
the "amica" board, esp12e+cp2104 module.

karlp, 2016-10-31:
Seen again with dev branch: 0.9.6-dev_20150627-911-g9db0778

```
es dbg: 	327: 	enduser_setup_connected_callback
Connected to wifi ssid: tweak our IP: 192.168.3.202   << user callback
es dbg: 	1137: 	enduser_setup_http_connectcb
es dbg: 	1021: 	enduser_setup_http_recvcb
es dbg: 	1137: 	enduser_setup_http_connectcb
es dbg: 	1021: 	enduser_setup_http_recvcb
es dbg: 	1137: 	enduser_setup_http_connectcb
es dbg: 	1021: 	enduser_setup_http_recvcb
es dbg: 	779: 	enduser_setup_serve_status
es dbg: 	696: 	enduser_setup_http_serve_header
es dbg: 	1137: 	enduser_setup_http_connectcb
es dbg: 	1021: 	enduser_setup_http_recvcb
es dbg: 	1526: 	enduser_setup_stop
es dbg: 	1211: 	enduser_setup_station_stop
es dbg: 	1400: 	enduser_setup_dns_stop
es dbg: 	1200: 	enduser_setup_http_stop
es dbg: 	1317: 	enduser_setup_free

-- some time passes
> 
> 
>  -- call start() again
> enduser_setup.start(es_ok, es_bad, es_dbg)
es dbg: 	340: 	enduser_setup_check_station_start
es dbg: 	1219: 	enduser_setup_ap_start
es dbg: 	1348: 	enduser_setup_dns_start
es dbg: 	1154: 	enduser_setup_http_start
es dbg: 	1163: 	http_start bind failed  << fails
es dbg: 	1526: 	enduser_setup_stop
es dbg: 	1211: 	enduser_setup_station_stop
es dbg: 	1400: 	enduser_setup_dns_stop
es dbg: 	1200: 	enduser_setup_http_stop
es dbg: 	1317: 	enduser_setup_free
>
```

marcelstoer, 2016-11-04:
> Seen again with dev branch: 0.9.6-dev_20150627-911-g9db0778

Sounds wrong to me. There was no EUS module in 0.9.6.

Try with #1583 and see if that changes anything.

karlp, 2016-11-08:
"dev branch" ie, git describe reports "0.9.6-dev_20150627-911-g9db0778" not the 0.9.6 release tag.

pulled 1583 in, and that's just worse.  AP doesn't even appear, and calling start twice results in a crash and reset.

```
> wifi.setmode(wifi.SOFTAP)
> enduser_setup.start(es_ok, es_bad, es_dbg)
es dbg: 	1597: 	enduser_setup_init: Debug callback has been defined
es dbg: 	1642: 	Performing AP Scan to identify likely AP's channel
es dbg: 	1489: 	enduser_setup_dns_start
es dbg: 	1230: 	enduser_setup_http_start
es dbg: 	409: 	enduser_setup_http_load_payload
es dbg: 	429: 	enduser_setup_http_load_payload unable to load file enduser_setup.html, loading backup HTML.
es dbg: 	437: 	Content is gzipped
es dbg: 	1263: 	enduser_setup_http_start info. Loaded backup HTML.
> =wifi.getmode()
2
> =wifi.SOFTAP
2
----- some time passes, AP never appears at all, try again....
> wifi.setmode(wifi.SOFTAP)
> enduser_setup.start(es_ok, es_bad, es_dbg)
es dbg: 	1628: 	enduser_setup_start
es dbg: 	1553: 	enduser_setup_init
es dbg: 	1557: 	init failed. Appears to already be started.
es dbg: 	1680: 	enduser_setup_stop
es dbg: 	1287: 	enduser_setup_ap_stop
es dbg: 	1541: 	enduser_setup_dns_stop
es dbg: 	1276: 	enduser_setup_http_stop
es dbg: 	1458: 	enduser_setup_free
es dbg: 	894: 	free_scan_listeners

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

marcelstoer, 2016-11-08:
> "dev branch" ie, `git describe` reports "0.9.6-dev_20150627-911-g9db0778" not the 0.9.6 release tag.

Argghh, `git describe` doesn't work well with lightweight (i.e. un-annotated) tags. Since we never tag `dev` anymore...you get it. Not nice.

> pulled 1583 in, and that's just worse. AP doesn't even appear, and calling start twice results in a crash and reset.

AP appears for me but I can confirm the initial problem and the reset.

marcelstoer, 2016-11-09:
Does someone with EUS know-how have time to look into this before the next master drop? Tagging @robertfoss @jfollas @jmattsson

robertfoss, 2016-11-09:
I'm short on cycles at the moment. Sorry.

jfollas, 2016-11-09:
Looking into it. Seems to be crashing if stop() is called (wasn't my
experience the other day, so something may have gotten munged during the
merge).

jfollas, 2016-11-19:
The issue with calling stop() without first calling start() just needed a NULL check for the internal state variable. I will submit a PR for this, but I don't think that's the issue that @karlp is reporting. Still trying to reproduce that.

jfollas, 2016-11-20:
Root cause of this issue seems to be trying to do a network scan without having the STA enabled. 

EUS performs a scan on startup to identify the AP with the strongest radio signal as part of a strategy to minimize radio channel switching by assuming that the user will probably want to connect to the closest AP. Since the STA and SoftAP both share the same radio, if the STA needs to change channels to connect to an AP, then the SoftAP will go along with it - result is that your phone won't be able to communicate with the SoftAP in that event until it times out and performs its own network scan to figure out what new channel to connect to. By starting the SoftAP on the most likely channel, there is [hopefully] no channel changing involved when the STA tries to connect to the AP for verification.

Added code to enable STA before the startup scan is performed so that it doesn't matter whether STA was previously enabled or not.

PR #1605 (2016-12-01): Eus bug fix
Fixes #1578.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

FIX: Ensure that station is enabled before doing network scan (this seems to be the cause of #1578)
FIX: Check if state == NULL on enduser_setup.stop()
IMPROVEMENT: Look for both enduser_setup.html.gz and endus [... cut]
closed 2016-12-01 by marcelstoer