nodemcu/nodemcu-firmware #1579: Generic buffer functionality

devsaurus, 2016-11-01:
### Missing feature
A generic buffer object model and derived variants for specialized applications.

### Justification
The `ws2812` module introduced a versatile buffer object for all kinds of data manipulation. This works pretty well and is optimized to achieve a high fps rate even for large LED chains and bitmaps. Other modules or programming tasks could benefit from a similar approach if they were able to operate on a buffer object instead of plain strings. My specific use case would be network buffers that are manipulated in-place instead of concatenating strings and so.

Since `ws2812` is continuously being improved (currently with #1575) but other modules are lagging behind, I'd like to start an architectural discussion to see whether a generic buffer object should be introduced in the firmware and how it would look like.

#### Generic buffer object
A buffer object is an array of 1 to `n` elements of 1 to `m` bytes in size. It's created dynamically by a module function and consumes a linear memory range of `n * m` bytes plus some overhead for Lua administration. The basic format for data exchange with agnostic Lua functions is a serialized string of this memory range.

We'd probably need a `buffer` module to offer the `new()` function.

##### Generic object methods
Many of the `ws2812` buffer methods are generic operations that can be established here as well, plus any additional ones:
- `get()`, `set()`
- `fill()`
- `size()`
- `shift()`
- `dump()`, `load()`
- `__tostring()`

#### Specialized buffer objects
A module like `ws2812` would offer a specialized "LED object" on top that adds methods which are meaningful in the context of LEDs:
- `fade()`
- `mix()`
- `power()`

I would be created by `ws2812.newBuffer()` like today. This can be achieved on C level by inheriting from the generic function map. Methods of the generic function map could also be overloaded by specialized variants.

We would need to establish some kind of shared thingy on C level so other LED modules can easliy use the same object model like in `apa102.newBuffer()` and `write()` that to the chain lateron.

### Workarounds
The same functionality can be achieved with string manipulation / concatenation, but this is inefficient in terms of code overhead and execution performance.

pjsg, 2016-11-01:
I support this. The existing buffer is good for the LED driving, but it isn't clear whether the "fixed shape" model is good for other uses.

pjsg, 2016-11-02:
Thinking about this some more, I'd like to let the ws2812 buffer thing settle down. Once people start to use it, then there will probably be some additional requirements. This would be a good time to move it off into its own buffer class.

devsaurus, 2016-11-02:
Agreed, the topics here should not block any development of ws2812 or others. Though unifying the LED modules was raised a couple of times in the past.
Once we have a better picture of what's needed, we can push towards an implementation.

closed 2019-06-21 by stale[bot]