nodemcu/nodemcu-firmware #1580: CJSON can not compare NULL

petrkr, 2016-11-01:
Based on comment in source code https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/cjson.c
I trying compare NULL from JSON response, but I always getting "false" even using suggested compare with cjson.null one.

### Expected behavior
return True

### Actual behavior
returns false

### Test code
> json_null_test = cjson.decode('null')
> print (json_null_test == cjson.null)
false

> print (json_null_test)
userdata: 0x00000000

> print (cjson.null)
nil

### NodeMCU version
NodeMCU 1.5.4.1+petrkr build 20161101 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32) based on NodeMCU - branch dev commit f0780e3

### Hardware
NodeMCU Devkit 1.0 with ESP-12

petrkr, 2016-12-25:
Found workaround for this issue:

```
cjson_null = cjson.decode('null') -- Define NULL object for compare 
json_null_test = cjson.decode('null') -- Parse real JSON
print (json_null_test == cjson_null) -- Compare with local variable
true
```

so I can use aswell in more complex JSON's

```
cjson_null = cjson.decode('null')
json_null_test = cjson.decode('{ "var1": "Test var string1", "null_var": null }')
print (json_null_test.null_var == cjson_null)
true
```

pjsg, 2017-01-23:
The new `sjson` module has an explicit `sjson.NULL` symbol which contains the null value in use.

closed 2017-02-18 by marcelstoer