nodemcu/nodemcu-firmware #1581: doesn't work with latest firmware in master when reseting the device

helxsz, 2016-11-03:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
I am using [NodeMCU v2 - Lua based ESP8266 development kit](https://www.seeedstudio.com/NodeMCU-v2---Lua-based-ESP8266-development-kit-p-2415.html#), and I used the latest firmware from [the automated nodemcu custom builds service](https://nodemcu-build.com/). I have tried to install the firmware in float and integer version. The custom firmware has the extra components like MQTT, I2C, 

### Expected behavior
When reseting the device, my ESPLORER, it should print out a few lines such as 

```
Can't autodetect firmware, because proper answer not received (may be unknown firmware). 
Please, reset module or continue.
�P��� ����d���
Restore init data.
� ��z����HF�Q��D

NodeMCU custom build  
	branch: master
	SSL: false
	modules: bit,file,gpio,i2c,mqtt,net,node,rtcmem,rtctime,spi,tmr,uart,wifi
 build 	built on: 
 powered by Lua 5.1.4 on SDK 1.4.0 
```

### Actual behavior

When reseting the device, my ESPLORER should print out a long line of random characters.
```
Can't autodetect firmware, because proper answer not received (may be unknown firmware). 
Please, reset module or continue.
`�  �-� ������� �=� ���;����� ��K$3���I�� ��@���;���� �-K@��� 1���`	 �8�@��K$3��� �=� ��+���`� �-K@��� ���$���-K@������� �-I@��� 1���	 �-I@��� �3�� �5� ���;���� �=� ��+���$�� �=� ���+�3���	 �� ���3���`� �=� ���;��� ���� ���3���` �-K@�����`� �8�@���;��� �� �=� ���+����  �5� ���;���$�� �5� ���;���$���-K@��+����  �-I@��+����� �5�@��I$3��� �� �=� ��+���I��  �8�@��I$;���� �� ���$3���$��	 �=� �� ���� �
� ��K$;�����8�@��I$;���I�� �=� ������ �=
```
and the nodemcu continues to blink the blue led, it looks like the nodemcu kit is getting busy and the ESPLORER will get stuck .

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
-- add code here
```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.
master branch
### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

pjsg, 2016-11-03:
Do you have the baud rate set to 115200 in esplorer?

helxsz, 2016-11-03:
@pjsg my previous baud rate is set to 9600,  all other baud rate fails except 74880. It repeatedly prints out 

```
load 0x40100000, len 24056, room 16 
tail 8
chksum 0xe3
load 0x3ffe8000, len 2260, room 0 
tail 4
chksum 0xfd
load 0x3ffe88d4, len 8, room 4 
tail 4
chksum 0x11
csum 0x11
rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

```

devsaurus, 2016-11-03:
The `rf_cal[0]` is an indication that calibration data is missing or wrongly stored.
See #1474 for similar symptoms. And, most importantly, check the link to flashing instructions that came from the build service.

helxsz, 2016-11-03:
@devsaurus , thanks. where can I find the esp_init_data_default.bin?

closed 2016-11-03 by helxsz

bernardmckenna, 2017-05-30:
holding D3 or GPIO0 high with a wire link works for me

jinfagang, 2017-10-08:
Hi, I setting both 9600 and 115200, but just got this error:
```
Can't autodetect firmware, because proper answer not received (may be unknown firmware). 
Please, reset module or continue.

```

Has no nodemcu info. Any help? I indeed flash the nodecmu firmware already build from nodecmu-build.com.