nodemcu/nodemcu-firmware #1582: MQTT get stuck in the connection?

helxsz, 2016-11-03:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
when mqtt connection fails, it will go to the failure callback function to help me identify the reason.

### Actual behavior
when mqtt connection fails, it will gets stuck and never get into the failure callback funciton, which is used to tell the failure reason.
### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
        print("trying connect to MQTT"..config.MQTT_HOST)
        
        connected = m:connect(config.MQTT_HOST, config.MQTT_PORT, 0,
           function(client)   
              print('mqtt connected, timer publisher ', timer_mqtt_publisher)      
               onMqttConnected()
            -- record online time
           end,
           function(client, reason)  // never call this function
                print("failed reason: "..reason) 
                print("Sleep for 10 secs")
                tmr.alarm(6,10000,tmr.ALARM_SINGLE, function()
                    conectMQTTBroker()
                end)
        end) 
```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.

NodeMCU custom build by frightanic.com
	branch: master
	commit: c8037568571edb5c568c2f8231e4f8ce0683b883
	SSL: false
	modules: bit,file,gpio,i2c,mqtt,net,node,rtcmem,rtctime,spi,tmr,uart,wifi
 build 	built on: 2016-06-01 19:59
 powered by Lua 5.1.4 on SDK 1.4.0

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

I am using NodeMCU v2 - Lua based ESP8266 development kit, and I used the latest firmware from the automated nodemcu custom builds service. I have tried to install the firmware in float and integer version. The custom firmware has the extra components like MQTT, I2C,

8<------------------------ END BUG REPORT -------------------------------------

devsaurus, 2016-11-03:
> build built on: 2016-06-01 19:59
> powered by Lua 5.1.4 on SDK 1.4.0

Please retry with a recent firmware build. There were quite a number of fixes in the `mqtt` module, e.g. #1445.

pjsg, 2016-11-04:
There is also the unmerged PR #1557 which deals with this case more closely.

closed 2016-11-08 by marcelstoer