nodemcu/nodemcu-firmware #1588: Unable to dim WS2812B

struthio, 2016-11-10:
### Expected behavior
> ws2812 is a library to handle ws2812-like led strips. It works at least on WS2812, **WS2812b**, APA104, SK6812 (RGB or RGBW).

WS2812 module can correctly interact with WS2812B Leds.

### Actual behavior

When I do:
```Lua
ws2812.init()
ws2812.write(string.char(255,0,0))
```
Then green light up.

```Lua
ws2812.init()
ws2812.write(string.char(255,255,0))
```
Then Yellow light up.

```Lua
ws2812.init()
ws2812.write(string.char(255,255,255))
```
Then WHITE is light up.

But there is no way of turning light off. When I do:
```Lua
ws2812.init()
ws2812.write(string.char(255,255,255))
ws2812.write(string.char(0,0,0))
```
Then LED stays White, this (0,0,0) has no effect.

Also codes like:
```Lua
ws2812.init()
ws2812.write(string.char(0,0,255))
```
Turn on GREEN led (works the same as (255,0,0)).

### NodeMCU version

NodeMCU is build from master and devel on NodeMCU Cloud Build Service.

### Hardware

I used ESP8266 in ESP-12F form and I used something like Image to shift levels between 3.3V and 5V.

pjsg, 2016-11-10:
Hmm. I just tried this, and it works for me, I directly connected the data line to the D4 pin on the nodemcu.

There are no substantive differences between dev and master for the 'write' method. 

In particular

```
ws2812.init()
ws2812.write(string.char(0,255,0))
```

causes my LED to turn red.

struthio, 2016-11-10:
Ar You using WS2812B or WS2812 ?

For me (WS2812B), Your code causes led to turn LED GREEN. For some reasons it seems like '0' are not transfered.

pjsg, 2016-11-10:
What happens with `ws2812.write(string.char(15,15,0))` ?

struthio, 2016-11-13:
Nothing happened. In fact until i reached something like 150 in value for color, led was blank.

I figured out what happened. I got incorrectly working level shifter 3.3V->5V. Now i fixed it and ws2812 module is working great.

Sorry for troubles.

closed 2016-11-13 by struthio