nodemcu/nodemcu-firmware #1590: Some simple operations are mind-bogglingly slow

pjsg, 2016-11-13:
### Expected behavior

Invoking `tmr.now()` should be quick.

### Actual behavior

It actually takes hundreds of microseconds!

### Test code

```Lua
a = tmr.now()
b = tmr.now()
print ('delay', b - a)
```

It turns out that finding `tmr` is very slow. The code

```Lua
now = tmr.now
a = now()
b = now()
print ('Delay', b - a)
```

works much better. However, it doesn't seem reasonable for everybody to do this for all module accesses. 

It seems that improving the lookups should be doable. (Spoiler alert: Yes, it is possible. The trick is to do it without burning RAM)

acdtrx, 2016-11-15:
i believe this is a lua issue or ... feature?

In most lua modules I know, outside nodemcu, it is customary to use things like:

```lua
local tmr = tmr

a = tmr.now()
```

for modules you plan to use in your file.

pjsg, 2016-11-15:
It turns out that there are all sorts of problems. Currently, I have tmr.now executing maybe 8 times faster.  There is some opportunity for improvement left. I'm learning about all sorts of aspects of the platform that I was unaware of!

pjsg, 2016-11-16:
Turns out that the byte read from flash fixup handler takes time (obviously). In order to read four bytes individually is 9.5us. To read them as a 32-bit read is 0.15us. This indicates that doing byte reads from flash is to be avoided whenever possible.

pjsg, 2016-11-17:
Things are now much better -- but the PR is a bit of a monster. 

The general approach was to make the ROTable lookups all be via a hash table. The has tables are computed at build time and injected very late with a bit of linker magic. This makes the lookups run in O(1) rather than O(n). Also fix the comparison functions used in the table search to not do any byte fetches -- thereby running faster. Also fix the build options so that all constant strings are 4 byte aligned so that accessing them is fast. 

If anybody has a lua benchmark that I can run on a stock nodemcu, then I'd love to try it.

TerryE, 2016-11-18:
Philip, I meant to post this yesterday but house-building got in the way.

The root issue is that the eLua ROTABLE patch is optimised for space rather than time.  It is useful to contrast the handling of regular globals with ROTable globals.  

-  First the regular globals uses a hash table and second it is based on a Lua string with is uniquely referenced in the string table so comparison is by 32 bit moniker.  
-  The ROTable use a linear scan in `luaR_auxfind()` which uses an O(N) `c_strcmp()` which runs like a double dog out of flash.

We have this hit for both the global lookup and the method lookup.

You mention patch and use the past tense, so I guess that you've been doing work on this already.  Is this linked to your other "ROMable compiled code" work?  We can either address this by coding practices (e.g. use locals to cache these resolutions, eg. `local now = tmr.now` or we have some more complex method of processing the image.  In this second case I would find it hard to imagine how we could do this with linker magic, but a two pass link might do the trick, that is do the main link and then post process to generate a supplementary overlay which is appended to the image. The two major components that it would need to contain are:

-   a RO Lua string table which enumerates all RO embed strings.  This would be an adjunct to the standard RAM string table.  String object addresses would be still unique and IIRC the eLua/ROTABLE patch in principle supports RO string objects but they aren't implemented. Any new strings would first be matched against the RAM table then the ROM table and only added to the RAM table on a double miss.  

-  a set of RO hash tables for ROTable resources.

-  doing this will have gone a long way to laying down some of the structures for your flash code caching so  you might consider adding this.

However testing this is going to be truly a bitch.  I really only managed to do my LCD patch because I could develop it and first test it using standard Lua (and the standard Lua test suite which threw up about half a dozen edge cases that I'd previously missed).  Once it was working under standard Lua, porting it was relatively trivial.

pjsg, 2016-11-18:
The current version that I have is divorced from the ROMable code work -- which I have shelved for a while. 

The version that I have consumes (I think) essentially no more RAM than the existing code. It does have to compare the strings from the RAM against the strings in the ROTable. However, it uses the hash of the RAM string to index into the tables. This means that essentially you only need one probe into the table. Then I have a custom memory compare that just does it with 32 bit reads (I also store the length of the ROM string so that I don't need to compute the length at runtime). 

I'd rather not add the strings to the global table even on a miss. The lookup is now *so much* faster that it would only be a small improvement (but it would burn RAM). 

The presence of the gdbstub was a boon. I have never managed to get the eLua stuff to build on linux -- which is a real pity. The ROMable code would really benefit from it.

Testing is not that bad -- it either works or it doesn't. There may be some corner cases where the old slow path code is taken. This is what I'm looking at now.

TerryE, 2016-11-18:
Philip,  the string handling in Lua is definitely a long way from optimal for the ESP32.  Take the `lvm.c` routine `l_strcmp()`.  This is really written assuming the target code will use H/W string manipulation routines.  Given that the lengths are known, then a simple `memcmp()` could be used or even (since strings are normally 32-bit aligned), an optimised version where they are aligned using 32 bit compare loop with a masked compare for the odd bytes.

(Sorry if you know all this but I've added some explanation for other readers).  All strings are currently stored in the hash `G(L)->strt` and  Lua string objects simple point to the corresponding unique entry in this table.  These are GCed pretty much like any other Lua object.  The Lua VM uses that fact are unique to collapse string comparison and assignment to using the object by reference.  This makes assignment and comparison run like shit of a shovel, but code like  `resp =  a .. b .. c ..d .. e` run like a dog singe all of the intermediate strings components have to be entered in the the string table and then GCed since they are nearly always used the once.

Anyway back to my suggestion.  It is that we have a RO table `G(L)->ROstrt` in the flashed firmware and the string objects are also stored in the firmware.  The change to support this in the VM is small since the table is really only referenced in string creation, and that is we only create a new string in `G(L)->strt` is it doesn't already exist in either table.   The ROtable is actually simplified since the ROtables are no longer linked cstring lists, but regular tables in the flash.   String assignment and comparison now longer needs special case handling.  References to `"tmr"` et al would no longer generate a new string in `strt`but simple pick up the copy in `RTstrt`.

The only real runtime change is that the GC would need to skip the RO entries.  

However what this does require is to be able to create these structures as compile / build time, and this would require a two-pass link / load algo, for example a first temporary image build which is then scanned to build a generated C module which contain all of the ROstrt and hash table static table declarations + initialisors which is then compiled and relinked into the final image.  OK, a little cludgy, but it would add a second or so to the build process but make the both runtime and RAM footprint smaller.

@jmattsson  have you got any thoughts on this as well as Phillip?

pjsg, 2016-11-18:
Interestingly enough, the memcmp/strcmp implementations in the ROM are actually quite good. It looks as though they are the hand crafted assembler versions from gcc/libc (can't recall which). 

What you propose is interesting -- I use the same approach (i.e. build the image and then scan over it to build the has tables, and then relink it with the new tables included) for the stuff that I did. 

I'm going to put up a PR -- not for merging, but to show the current state of the WIP.

TerryE, 2016-11-18:
Philip, I've been thinking about this some more whilst barrowing the odd 5 tonne of hardcore from the front of the house to level up round the back.  I had considered an alternative approach of the judicial use of macros during compile so that all of the storage structures are preallocated for linking, but not initialised.  We could then postprocess the bin file + the map to patch in the hash tables, etc. into the bin file.

However we couldn't use this approach to preallocate the ROstrt hash, which is a  problem. 

I think that this will be faster than your WIP ROtable patch but only by a percentage rather than the 10× that you've already achieved.  It will also help with RAM usage as strings for `tmr`, etc. won't need to be created.

jmattsson, 2016-11-18:
Question: precisely which string constants are we talking about here? If it's the module names and key entries only, then I could see it being relatively easy to set up dependencies that grep for NODE_MODULE and LSTRKEY, pulls the strings and generates a .c file with the hash table definition. Possibly with dependencies to build a host-version of the hasher to generate said .c file. If it's more strings, or it gets too difficult to avoid bloating the ROstrt, then it's less straight forward I suppose.

Doing a double-link is sub-ideal, but might be worth it. The speed increases mentioned here seem like we should keep the option on the table regardless. I'm not spending much time around the ESP8266 at the moment, but I believe whatever approach you guys come up with there, I should be able to replicate in the IDF for the ESP32.

@pjsg Gcc does have [optimized builtins](https://gcc.gnu.org/onlinedocs/gcc/Other-Builtins.html), yes. Just don't look at the x86_64 versions of some of them - they're monsters...
@TerryE You're so hardcore... ;)

TerryE, 2016-11-19:
Johny, in this first incarnation it would be mainly the module, method and constant names. The hash tables are keyed on the Lua interned string hash. These saving s (especially the time ones) would carry across onto the ESP32. I'll review P's patch and pick up the thread when I've done so.

tjuerges, 2017-04-22:
@TerryE 
> (since strings are normally 32-bit aligned)

Terry, can you point me to where that is enforced? Thanks!

TerryE, 2017-04-22:
@tjuerges Thomas, what I said wasn't quite correct.  There are some undocumented characteristics of the xtensa gcc backend.  With **-Os** strings are collated and condensed to byte alignment,  With **-O2** that are long (4-byte) aligned.  It's not a Q of enforcement, but a behaviour of code generator.

tjuerges, 2017-04-22:
@TerryE That is very valuable information.  Thanks for clarifying!

closed 2019-06-21 by stale[bot]