nodemcu/nodemcu-firmware #1592: file.read() causes reboot

Gatulibu, 2016-11-16:
According to the documentation file.read() should:

> if nothing passed in, read up to LUAL_BUFFERSIZE bytes (default 1024) or the entire file (whichever is smaller)
> 
> Returns
> File content as a string, or nil when EOF

But when I try:

```
file1 = file.open("file1k5") -- 1.5kB file
print(file1.read())  -- prints first 1024 B
print(file1.read())  -- prints remaining data
print(file1.read())  -- does not return nil but causes a reboot "cause:2"
file1.close()
```

The same happens on an empty file:

```
file1 = file.open("empty1", "w")  --create empty file
file1.close()

file2 = file.open("empty1")
print(file2.read())  -- module reboots
file2.close()
```

 
 Version used:

>  NodeMCU custom build by frightanic.com
> 	branch: dev
> 	commit: 798759950fe03976204430fc6a4a64fffa04f34a
> 	SSL: true
> 	modules: adc,crypto,file,gpio,i2c,net,node,pwm,struct,tmr,uart,websocket,wifi,ws2812
>  build 	built on: 2016-11-13 18:46
>  powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

jmattsson, 2016-11-16:
When using the returned file object, you need to either use the `:` syntactic sugar (recommended), or explicitly pass the file object as the first argument. I.e.
```
file1 = file.open("empty1")
print(file1:read())
file1:close()
```
or
```
file1 = file.open("empty1")
print(file1.read(file1))
file1.close(file1)
```

This is the way Lua works with "objects", it's not something NodeMCU specific. See e.g. https://www.lua.org/pil/16.html

Gatulibu, 2016-11-17:
Thank you for clarifying the correct use of the new object model for files (which was recently merged into the dev branch #1532).

 
Unfortunately I have to report that both suggested ways fail the same way as before (dev branch).

 
In the master branch the following commands work without a problem:

```
file.open("empty1", "w")
print(file.read())
file.close()
```

In the dev branch I get a reboot at the read command.

As previously described a read on a non empty file (after EOF has been reached) also causes a reboot.

```
_flash_used_end:402639c3
fs.start:70000,max:38a000
_flash_used_end:402639c3
fs.start:80000,max:37a000
mount res: 0, 0
Task task_lua started.
nul mode, fpm auto sleep set:enalbe

NodeMCU custom build by frightanic.com
	branch: dev
	commit: 1d0cf5e8d050ab1970593c9306d356bf70abfdf5
	SSL: false
	modules: file,gpio,net,node,tmr,uart,wifi
 build 	built on: 2016-11-16 22:19
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
lua: cannot open init.lua
> Heap size::44320.
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with wifi123, channel 13
dhcp client start...

	wifi_event_monitor_handle_event_cb is called
ip:192.168.2.31,mask:255.255.255.0,gw:192.168.2.1

	wifi_event_monitor_handle_event_cb is called
pm open,type:2 0
f1 = file.open("empty1", "w")  --create empty file
> f1:close()
> f2 = file.open("empty1")
> print(f2:read())
Fatal exception 28(LoadProhibitedCause):
epc1=0x402475d7, epc2=0x00000000, epc3=0x00000000, excvaddr=0x1ffef66c, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 25864, room 16 
tail 8
chksum 0x14
load 0x3ffe8000, len 2200, room 0 
tail 8
chksum 0x7b
load 0x3ffe8898, len 8, room 0 
tail 8
chksum 0x28
csum 0x28

```

devsaurus, 2016-11-17:
The symptoms look strange, I'll have a look later that day.

Gatulibu, 2016-11-19:
I can confirm that #1599 fixed the issue. Thank you.

PR #1599 (2016-11-19): Handle error condition in file.read()
Fixes #1592.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

SPIFFS reports an error code when reading from an empty file (why not result in 0 len read instead?). This was not caught up to now in `file.read()` and led the firmware to crash.
Any error conditions are now handled and transformed into an empty read aka `n [... cut]
closed 2016-11-19 by Gatulibu