nodemcu/nodemcu-firmware #1593: More documentation on net.socket:hold() would be helpful

marcelstoer, 2016-11-16:
Triggered by a question at http://www.esp8266.com/viewtopic.php?f=23&t=12540 I found [`net.socket:hold()`](http://nodemcu.readthedocs.io/en/latest/en/modules/net/#netsockethold). I must agree that the documentation is every but clear. A few vital pieces of information seem to be missing.
That thread also links to a question posted to an Espressif forum by @nkolban asking for the same kind of clarification. He never got a reply.

Does one of the developers who contributed to the net module have more info?

devsaurus, 2016-11-17:
I don't have further links, but used the hold and unhold functions in the [PCM network streaming example](https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_examples/pcm/play_network.lua). They work "as expected" and notify the server to stop sending packets until further notice. It can stream several Mbytes of audio this way.
They're essential in cases when the data source (http server) is faster than the data sink (16k audio). RAM buffers would very quickly overflow otherwise.

jmattsson, 2016-11-17:
Are they really essential? The TCP receive window would the throttling the flow regardless.

devsaurus, 2016-11-17:
The TCP windowing would throttle, I agree. But I couldn't find any other solution on Lua level:

My code has to fetch the data when the receive cb fires and eventually buffer it before it's fed into the audio backend. That frees up net and TCP layer, so they happily accept more data from the sender and spill it through the next receive cb. I can't block the cb per se in order to propagate the full-stop downwards the networking stack.

In the end it's my Lua app that gets blown up in case I can't throttle the sender according to my bandwidth constraints.

marcelstoer, 2016-12-02:
> They work "as expected" and notify the server to stop sending packets until further notice.

closed 2016-12-31 by marcelstoer