nodemcu/nodemcu-firmware #1595: dev-esp32 firmware module fail!!

aesirteam, 2016-11-17:
NodeMCU ESP32 build unspecified powered by Lua 5.1.4 on IDF 1.0.0(5513faf)
lua: cannot open init.lua
> 
=1+2
=1+2
3
> Waiting answer from ESP - Timeout reached. Command aborted.
stdin:1: attempt to index global 'node' (a nil value)
> =node.info()
stdin:1: attempt to index global 'node' (a nil value)
> =node.chipid()
stdin:1: attempt to index global 'node' (a nil value)
> =node.flashid()
stdin:1: attempt to index global 'node' (a nil value)
>

jmattsson, 2016-11-17:
I can confirm that there seems to be an issue where none of the optional modules get linked in. Presumably this is a result of the recent IDF build system changes; I'll have a look when I get home tonight.

jmattsson, 2016-11-17:
Actually, that's a trivial one-line fix, let me do that while I'm compiling other stuff here...

Okay, fixed - thanks for reporting! :)

closed 2016-11-17 by jmattsson