nodemcu/nodemcu-firmware #1597: Fail node.restart by dev-esp32

aesirteam, 2016-11-17:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
NodeMCU broad restart
### Actual behavior
NodeMCU broad can restart, but stop (or waitting?  error?)  
### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
node.restart()
```
### NodeMCU version
dev-esp32

### Hardware
NodeMCU ESP32

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

aesirteam, 2016-11-17:
debug mode:
=node.restart()
=node.restart()
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x0  1 panic'ed (  1 panic'ed (  1 panic'ed (  1 panic'ed (  1 panic'ed (  1 panic'ed (  1 panic'ed (  1 panic'ed (  1 panic'ed (  1 panic'ed (  1 panic'ed (

jmattsson, 2016-11-17:
This is a known issue in the IDF, see https://github.com/espressif/esp-idf/issues/86

closed 2016-11-17 by marcelstoer

jmattsson, 2016-11-24:
FYI, I just pushed the IDF update which fixes this.