nodemcu/nodemcu-firmware #1598: net.socket:on "receive" fails to manage payload chunking for encrypted connection

RadianM, 2016-11-17:
### Expected behavior
Having established a secure connection to a webserver using net.createConnection with the encrypted flag set, a GET request should return an arbitrarily large payload in a series of receive callbacks (usually delivering multiples of 1460 bytes)

### Actual behavior
Receive Payloads above a critical size (somewhere between 5K~6K bytes) variously generate zero, one or two receive callbacks (in addition to the header which always comes in OK) - terminating with the error "client's data invalid protocol". Differing behaviour can be obtained with successive identical requests. Payloads below the critical size are returned (seemingly always in one callback) without any errors.

### Test code
Finding a public test server is a little challenging but https://pci-test.cyssecure.com/ will return the results of secure access attempts using TLS1.1 so I have hard-coded the request as follows:
```Lua
wifi.setmode(wifi.STATION)
wifi.sta.config("ssid","pass")
wifi.sta.connect()

tmr.alarm(0, 100, 1, function()  
    if wifi.sta.getip() ~= nil then
        tmr.stop(0)
        sendRequest()
    end
end)

function sendRequest()
    svr=null
    svr = net.createConnection(net.TCP, 1)  --SSL=1
    svr:on("connection", function(skt)
        print("connected to server. Heap="..node.heap())
        skt:send("GET / HTTP/1.1\r\nHost: pci-test.cyssecure.com\r\nConnection: close\r\nAccept: */*\r\n\r\n")
        print("> Sending Request. Heap="..node.heap())
    end)
    
    svr:on("reconnection", function(skt)
        print("> reconnection. Heap="..node.heap())
    end) 

    svr:on("sent", function(skt)
       print("> Request sent. Heap="..node.heap())
    end) 

    svr:on("receive", function(skt, payload)
        print("> rx payload len="..string.len(payload).." Heap="..node.heap())
        print(payload)
    end)

    svr:on("disconnection", function(skt)
        print("> disconnection. Heap="..node.heap())
        skt:close()
    end)    
    svr:connect(443,"pci-test.cyssecure.com")
end
```
While the above request consistently fails, I have yet to be able to find a public server that delivers a small enough payload to demonstrate a successful session. In my own development environment I am making secure requests to a LED Lighting gateway that has a variable sized payload depending on the number of LED lamps connected. I had no indication of any problems until one day I added an extra lamp and my nodemcu-based home automation system failed.

### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 1d0cf5e8d050ab1970593c9306d356bf70abfdf5
	SSL: true
	modules: file,gpio,net,node,tmr,uart,wifi
 build 	built on: 2016-11-16 11:07
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
(Debug enabled)

### Hardware
LoLin V3 NodeMcu

marcelstoer, 2016-11-17:
> I have yet to be able to find a public server that delivers a small enough payload to demonstrate a successful session.

The http://httpbin.org/ endpoints wouldn't help, right? IIRC they don't work over HTTPS for NodeMCU (TLS version and/or cipher suites). Maybe https://requestb.in?

Since you communicate over HTTP did you try the NodeMCU HTTP module? It uses an internal [8k buffer](https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/http/httpclient.h#L40).

RadianM, 2016-11-17:
Unfortunately neither of those services get past the client handshake. However I just found that https://www.example.net returns a small payload. So replacing these two lines in my function above:
`skt:send("GET / HTTP/1.1\r\nHost: example.net\r\nConnection: close\r\nAccept: */*\r\n\r\n")`
`svr:connect(443,"example.net")`
yields one 1270 byte callback in addition to the header. Having said that, an error is thrown following the receipt of the payload: "client's data invalid protocol. Error: SSL error 3". I don't see this error ( ‘SSL certificate name mismatch’? ) after successful requests from the Lighting gateway.

I actually started out with the HTTP module but the single callback with fixed buffer size didn't scale with my application. Arbitrarily large payloads should be no problem if the firmware correctly handshakes it in for successive callbacks.

marcelstoer, 2017-03-09:
Do you still see this with v2.0 (new net module and mbedTLS)?

closed 2019-06-21 by stale[bot]