nodemcu/nodemcu-firmware #1602: SPI: can't setup spi.CPOL_LOW, spi.CPHA_HIGH

rptnscz, 2016-11-18:
### Expected behavior
setup spi

### Actual behavior
hangs, reboot

### Test code
spi.setup(0, spi.MASTER, spi.CPOL_LOW, spi.CPHA_HIGH, 32, 16, spi.HALFDUPLEX)

### NodeMCU version
Nodemcu 1.5.4.1

### Hardware
ESP8266-12, Nodemcu Devboard, https://www.aliexpress.com/item/Update-Industry-4-0-New-esp8266-NodeMCU-v2-Lua-WIFI-networking-development-kit-board-based-on/32358722888.html

luismfonseca, 2016-11-19:
Have you read the recently added note on the [SPI](http://nodemcu.readthedocs.io/en/dev/en/modules/spi/) module documentation?

Can you try this using HSPI or ID 1? It doesn't seem to crash if you're using `id = 1`.

rptnscz, 2016-11-19:
Thank you for your answer.  In the meantime, I've tested spi.setup(0, spi.MASTER, spi.CPOL_LOW, spi.CPHA_HIGH, 32, 16, spi.HALFDUPLEX) on the adapter plate welding ESP-07 chip - it behaves the same: hangs, reboot.  https://www.aliexpress.com/item/1set-ESP8266-serial-WIFI-module-adapter-plate-Applies-to-ESP-07-ESP-08-ESP-12/32381092806.html

The note explains the problem well and describes the solution, thank you! I haven't noticed it, sorry. And yes, spi.setup with HSPI works like a charm on both devel board as well as adapter plate+ESP-07. But I have not tested SPI communication  yet, looking forward to do it in the next few days.

Thank you for your help,
--Roman

FrankX0, 2017-01-02:
I guess this issue is solved, correct?
Can you close?

rptnscz, 2017-01-02:
Unfortunately, I still haven't hardware I would like to test with. Aliexpress Christmas delay or just lost package. 
Anyway I believe it is solved. Please, close the issue. 

Thank you for your help, 
Roman

closed 2017-01-02 by marcelstoer