nodemcu/nodemcu-firmware #1607: How is the Lua C integration of the current apps done?

joakimeriksson, 2016-11-22:
### Missing feature
Documentation on how to add a C-module - or rather some examples of how the current apps are integrated into LuA. How do Lua understand how to use CoAP? I did not find any Lua binding information in the CoAP code - just some bits and pieces in endpoint.c but nothing else. Is there more
files that are not part of the app directory or is there a magic auto-wrapper that does it?

### Justification
I am considering to add some of our app from Contiki OS into NodeMCU such as LWM2M client and I feel that I need to understand this better first.

marcelstoer, 2016-11-22:
We have some information in https://github.com/nodemcu/nodemcu-firmware/wiki/%5BDRAFT%5D-How-to-write-a-C-module

joakimeriksson, 2016-11-22:
Yes I did read that one and then I tried to use that knowledge to figure out how for example the CoAP module was supported in Lua - but I did fail to find the binding information in the coap code. Is it developed in the same way or is there another principle that was used for the coap app?

marcelstoer, 2016-11-22:
Not sure I can really help you as I'm not a firmware programmer but I'll try anyway. The Lua binding is in [`/app/modules/coap.c`](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/coap.c). The heavy-lifting is done in [`/app/coap`](https://github.com/nodemcu/nodemcu-firmware/tree/master/app/coap). Lots of modules have their own subfolder in `/app` and then refer to that from within `/app/modules/<module>.c`.

joakimeriksson, 2016-11-22:
Yes, that is absolutely the help I needed - I was obviously a bit blind! Thanks!

closed 2016-11-22 by TerryE