nodemcu/nodemcu-firmware #1608: Memory leakage in net.sock with internet connection problems

Modellfan, 2016-11-22:
8<------------------------ BUG REPORT -----------------------------------------
### Memory-Leakage free net.socket

### Memory-Leakage when internet connection is unstable

### Test code
I'm not a pro in Lua, but I already spent a lot of time to write a stable http get function for thingspeak.
My nodemcu device has sometimes a bad Internet connection, because it is connected to repeater
and the repeater is connected to the router. When connection is good the code is leagage free, if 
not it leaks around 1kb each call. I broke down my code to the very basic to find out the reason for this
behaviour, but I got to a point where I believe , the problem is either lua firmware or sdk. Btw
thank you guys for the work on the lua firmware.

This code calls my getter function:
```Lua
tmr.alarm(0, 1000, tmr.ALARM_AUTO, 
function()
  print("---------------------")
  print("Heap vorher:"..node.heap())
  local t = tmr.now()
  dofile("getter.lua")(
  function()
    --tmr.start(0)
    print("Latency: "..(tmr.now()-t)/1000000)
    print("Recalled")
  end)
  collectgarbage()
  print("Heap nachher:"..node.heap())
end)
```

And this is the getter, I derived from a post I found on github, with outcommented timeout function:
```Lua
return function (retfunc)

local connected = false
local conn = net.createConnection(net.TCP, 0)

conn:on("connection", function(c)
   c:send("GET / HTTP/1.1\r\n\r\n")
   print("Connected")
   connected = true
end)

conn:on("receive", function(c,pay)
   --print(pay)
   print("Recieved")
end)

conn:on("disconnection", function(c)
   c:close()
   tmr.unregister(6)
   
   conn = nil
   timeout = nil
   collectgarbage()
   print("Disconnected")
   retfunc()
end)

conn:connect(80, '184.106.153.149')

local timeout = function(c)
  print("Timed out!")
  if connected then
    --hierhin kommen wir falls Verbindung etabliert aber keine Antwort
    --sollte auch dissconnect aufrufen und da aufräumen
    c:close()
  else
    --hierhin kommen wir falls NodeMCU kein Internet hat
    --wir müssen direkt hier aufräumen, weil disconnect nicht aufgerufen wird
    c:close()
    conn = nil
    collectgarbage()
  end

  --it is not necessary too nil something because this calls 
  --the disconnect routine
end

--tmr.alarm(6, 1000, tmr.ALARM_SINGLE, function() timeout(conn) end)

return nil

end

```
### NodeMCU version
Dev 18

### Hardware
Nodemcu with nothing attached

8<------------------------ END BUG REPORT -------------------------------------

marcelstoer, 2016-11-26:
Any particular reason you're not using [`http.get()`](http://nodemcu.readthedocs.io/en/latest/en/modules/http/#httpget)?

Modellfan, 2016-11-28:
Yeah, because http.get() also produced memory leak in this case and i went one level down to find out the particular reason.

sinedied, 2017-01-10:
I also experience memory leak with the `net` module.
My code is somewhat different: https://github.com/sinedied/nodemcu-experiments/blob/master/rc.lua

Even reducing the server to
```lua
srv = net.createServer(net.TCP) 
srv:listen(80, function(conn) 
  conn:on("receive", function(conn,payload)
    conn:close()
    collectgarbage()
    print(node.heap())
  end)
end)
```
leaks over time.

marcelstoer, 2017-01-10:
@sinedied this is expected to leak because you're reusing the `conn` variable in the 2nd callback. Stick to the example in the documentation at http://nodemcu.readthedocs.io/en/latest/en/modules/net/#example_3.

sinedied, 2017-01-10:
@marcelstoer Thanks for pointing my mistake, bu even after fixing this like in the example there is still leaks after every connection :/

marcelstoer, 2017-01-10:
@sinedied This issue should actually be closed because the initial code is far from being an  [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve). I doubt this issue will ever make progress in its current shape or form. Even less now as with your example it's taking a different direction. 

This is not to say though that there aren't any leaks in the net module. I can only partially confirm your observations. I used the example from the documentation and then sent a few requests to it using `echo "foo" |nc 192.168.0.46 80`. It's true that available heap takes a hit with every connection but it _eventually_ bounces back even though running `collectgarbage()` before that doesn't seem to help. It takes some 10s of seconds or maybe 1 minute which _may_ indicate hanging sockets/connections. 
I was testing on current `dev` which contains the re-writen net module. @djphoenix any ideas?

djphoenix, 2017-01-11:
@sinedied @marcelstoer I think there is not a leak but kept connections in `TIME_WAIT` state (as we noted [there](https://github.com/nodemcu/nodemcu-firmware/pull/1379#issuecomment-268074953)). Maybe we need to tune LWIP configurations...

marcelstoer, 2017-01-11:
Thanks Yury, I remember that conversation vaguely. However, I fail to understand where lwIP would be waiting. After all, the socket is manually closed for every client even though the listening socket remains active. For the sake of completeness here's the example from the documentation in plain sight:

```lua
sv = net.createServer(net.TCP, 30)

function receiver(sck, data)
  print(data)
  sck:close()
end

if sv then
  sv:listen(80, function(conn)
    conn:on("receive", receiver)
    conn:send("hello world")
  end)
end
```

djphoenix, 2017-01-11:
> However, I fail to understand where lwIP would be waiting.
This is not for LWIP but for TCP. After connection closed, socket remaining (system-wide) in state called `TIME_WAIT` for handle last frames that may be (re)sent by remote side. This time is near 60s if I remember it right. In *nix systems you can see it by:
```
sudo netstat -46nt | grep TIME_WAIT
tcp6       0      0 192.168.97.1:80         192.168.97.36:55397     TIME_WAIT  
tcp6       0      0 192.168.97.1:80         192.168.97.36:55641     TIME_WAIT  
```
So `espconn` drops them right after socket closed.

> After all, the socket is manually closed for every client even though the listening socket remains active.
For TCP, "listening" socket and "connection" socket is not same. Listening socket only waits for incoming connections (SYN frame), and when receiving it, creates a new "connection" socket between endpoints.

marcelstoer, 2017-01-11:
> For TCP, "listening" socket and "connection" socket is not same. Listening socket only waits for incoming connections (SYN frame), and when receiving it, creates a new "connection" socket between endpoints.

Yep, I knew that. It's that "connection" socket that is closed with `sck:close()`.

> After connection closed, socket remaining (system-wide) in state called `TIME_WAIT` for handle last frames that may be (re)sent by remote side. 

Thanks, I didn't know that but I'll read up on it. It explains the behavior I observed with my tests.

closed 2017-01-11 by marcelstoer