nodemcu/nodemcu-firmware #1610: PANIC: unprotected error in call to Lua API   (esplorer)

bicapel, 2016-11-26:
Hello,
I am doing a project. ESP says:
PANIC: unprotected error in call to Lua API (stdin:14: attempt to call global 'is_buz_on' (a boolean value))

Is there a way to avoid it?

Here is my ESP code:

BUZ_PIN  = 6 -- GPIO 12

FIRE_PIN = 7 -- GPIO 13

GAS_PIN  = 5 -- GPIO 14
 
BUZZ_TMR = 1
FIRE_GAS_TMR = 2

function is_buz_on()
    running, mode = tmr.state(BUZZ_TMR)
    return running
end

function buz_on()
    is_buz_on = false
 
    tmr.alarm(BUZZ_TMR, 1000, tmr.ALARM_AUTO, function()
        if is_buz_on then
            gpio.write(BUZ_PIN, gpio.LOW) -- on
        else
            gpio.write(BUZ_PIN, gpio.HIGH) -- off
        end
 
        is_buz_on = not is_buz_on
    end)
end
 

function buz_off()
    gpio.write(BUZ_PIN, gpio.HIGH)
    tmr.unregister(BUZZ_TMR)
end
 
 

function notify_pushbullet()
    http.post('https://api.pushbullet.com/v2/pushes',
              'Content-Type: application/json\r\nAccess-Token: TOKEN_DO_PUSHBULLET\r\n',
              '{"body":"O sensor detectou fogo ou gás, verifique.","title":"Fogo detectado!!!","type":"note"}',
    function (code, data)
        print(code)
    end)
end
 

function start()
    buz_off()
 
    tmr.alarm(FIRE_GAS_TMR, 500, tmr.ALARM_AUTO, function()
        turn_buz_on = false
 
        -- verifica se detectou chamas.
        if gpio.read(FIRE_PIN) == 1 then
            turn_buz_on = true
        end
 
        -- verifica se detectou gás.
        if gpio.read(GAS_PIN) == 1 then
            turn_buz_on = true
        end
 
        if turn_buz_on then
            if not is_buz_on() then
                buz_on()
                notify_pushbullet()
            end
        else
            buz_off()
        end
    end)
end
 

function stop()
    buz_off()
    tmr.unregister(FIRE_GAS_TMR)
end
 
function configure()
    -- configura os GPIO.
    gpio.mode(BUZ_PIN, gpio.OUTPUT)
    gpio.mode(FIRE_PIN, gpio.INPUT)
    gpio.mode(GAS_PIN, gpio.INPUT)
 
    -- configura o ESP para conectar no Wifi.
    wifi.setmode(wifi.STATION)
    wifi.sta.config("ssid", "pass")
    wifi.sta.connect()
end
 
configure()
start()

dnc40085, 2016-11-26:
from  http://nodemcu.readthedocs.io/en/master/en/support/

> The issues list on GitHub is not the right place to ask for help. Use it to report bugs and to place feature requests. Questions like "how do I ..." or "I can't get this to work ..." should be directed to [stackoverflow.com](http://stackoverflow.com/) or [esp8266.com](http://www.esp8266.com/viewforum.php?f=17)

jmattsson, 2016-11-27:
Closed as per #1010.

closed 2016-11-27 by jmattsson