nodemcu/nodemcu-firmware #1613: ESP32: rewrite file module

jmattsson, 2016-11-27:
**Rationale:**
Much of the filesystem support currently in NodeMCU32 needs to be tossed in favour of the native IDF functionality. The earlier we can do that, the less maintenance effort for us. Things needing to change:

 - [ ] Rewrite `file` module to use standard C file functions. The IDF provides VFS support internally.
 - [ ] Remove the NodeMCU platform/vfs layer, since it's not needed with the IDF.
 - [ ] Reimplement `file.mount()` in terms of IDF VFS handling.
 - [ ] Support only object-based file API. There is no reason to support the legacy approach here; the overhead from a userdata object compared to extra Lua-level logic to deal with files sequentially should be on par at worst.
 - [ ] Consider renaming the `file` module to `fs`, since the operations left on it are less file related and more file *system* releated.

The two key reasons for moving onto the IDF VFS / standard C interface are:
 - It avoids code duplication -> smaller resulting binary
 - Flash encryption. Using native IDF components will mean a lot less work for us, compared to having to rewrite our stuff to handle the encrypted scenario (where all flash I/O has to be mmap()'ed, effectively)

**Priority:** High

**Dependencies:**
  - Native IDF SPIFFS being released (so far only the partition type is out). If this takes longer than expected, the interrim approach would be to just wire our SPIFFS into the IDF VFS, which would allow most of the above work to be done.

**Note:**
We should probably change our primary partition type to match the new IDF pre-defined SPIFFS partition type.

LDmitryL, 2017-03-28:
@jmattsson When working with the module file, I noticed that this strange behavior of the operator `file.readline()`. The documentation says that `LUAL_BUFFERSIZE` equal to `1024` bytes (by default). But when reading a long string (200 bytes) reads only 128 bytes, and the next time the rest of 200-128=72 bytes. How to set the buffer size as default?

jmattsson, 2017-03-28:
At some point LUAL_BUFFERSIZE got changed to be BUFSIZ, but a quick grep suggests that BUFSIZ should indeed be 1024. Not sure what's going on here, but it sounds like it needs some investigation (not sure when I'll find time however).

LDmitryL, 2017-03-28:
@jmattsson It seems to me that there are deeper consequences.
Do the IDF in the file stdio.h there is a define `BUFSIZ -> 1024.`
But there is another definition of `BUFSIZ  ->  __BUFSIZ__.`
And in the file config.h under the comment `/* ESP8266 - specific ... */` 
define `__BUFSIZ__ -> 128.`
Perhaps somewhere there is still confusion with the adaptation ESP8266 under ESP32.

LDmitryL, 2017-03-28:
It is a serious problem, maybe!?
Maybe you want to publish it in issue?

jmattsson, 2017-03-29:
I cannot find the `__BUFSIZ__` -> `128` setting in any `config.h` file (or any other file) in my repo. Where did you spot that?

jmattsson, 2017-03-29:
I've pushed a change to put LUAL_BUFFERSIZE back to 1024, but I think we should clean up the `file` module to not be so tightly coupled to LUAL_BUFFERSIZE - it would be advantageous to not instantiate such large buffers all the time. Something for someone who has time however...

LDmitryL, 2017-03-29:
Well if the size of the buffers can be set before opening file.
Something like this `file.buffer(512)`
And now what I need to do? I need a buffer of 1024.

jmattsson, 2017-03-29:
Yeah dynamically sizing the buffer depending on the requested length would be much better.

Right now, if you pull the latest `dev-esp32` you'll get the 1024 buffer size.

LDmitryL, 2017-03-29:
Somewhere shot down by logic and definitions can be so:

_in file nodemcu-firmware\components\lua\luaconf.h_
`#define LUAL_BUFFERSIZE		BUFSIZ`

_in file nodemcu-firmware\tools\toolchains\esp32\xtensa-esp32-elf\include\stdio.h_
`#ifdef __BUFSIZ__`
`#define	BUFSIZ		__BUFSIZ__`
`#else`
`#define	BUFSIZ		1024`
`#endif `

_in file nodemcu-firmware\sdk\esp32-esp-idf\components\newlib\include\sys\config.h_
`#ifdef __XTENSA__`
`#include <xtensa/config/core-isa.h>`
`#define MALLOC_ALIGNMENT ((XCHAL_DATA_WIDTH) < 16 ? 16 : (XCHAL_DATA_WIDTH))`
`/* esp8266-specific: shrink the default fd buffer size */`
`#define __BUFSIZ__ 128`
`#ifndef __DYNAMIC_REENT__`
`#define __DYNAMIC_REENT__`
`#endif`
`#ifndef _REENT_SMALL`
`#define _REENT_SMALL`
`#endif`
`#define HAVE_GETOPT`
`#endif`

jmattsson, 2017-03-29:
`luaconf.h` was changed in e33fb81b7780f6d2a380b36d8ac95337f4f4080d.

devsaurus, 2017-04-17:
I've set up an `sdmmc` module over at https://github.com/devsaurus/nodemcu-firmware/tree/dev-esp32-fatfs that will prepare the sdmmc peripheral to talk to an external SD card. Implements raw sector read & write functionality and also mounting the filesystem.
This is still under the umbrella of the NodeMCU platform/vfs with only minor changes to the `file` module (mount() removed).

I regard this as just one piece of the puzzle to move to a `fs` module, as the points in the original post are still valid:
- move to IDF's vfs tbd
- no native support for SPIFFS in the IDF, eventually craft our own
- all enhancements for `file` still pending, like oo-API and multiple open files

jmattsson, 2017-05-04:
I've started looking at this. It seems to me we should simply bring in the official Lua `io` module (sans `popen()` and `tmpfile()`), and then have a new `fs` module to deal with mount/unmount/format/rename/unlink/fstat/ls type operations.

jmattsson, 2017-05-07:
I finally got to spend a few hours on ESP32 stuff, so I've started work [here](https://github.com/jmattsson/nodemcu-firmware/tree/dev-esp32-fs-io-fat). Very much work-in-progress so far. `io` module is back, but only FATFS+WL hooked up the VFS so far. SPIFFS and plain FATFS to come...

joysfera, 2017-06-28:
I hacked together few bits to get spiffs for esp32. If you were interested -> https://github.com/joysfera/esp-idf-spiffs

closed 2019-07-12 by stale[bot]