nodemcu/nodemcu-firmware #1615: ESP32: finalise wifi module

jmattsson, 2016-11-27:
**Rationale:**
The `wifi` module is in a minimalistic state at this point, and sorely undocumented. Since the IDF API changed compared to the old 8266 SDK, I took the opportunity to start afresh with this module, incorporating much of the feedback from @marcelstoer in earlier discussions around the `wifi` module API in terms of semantics and naming. More work is needed to finalise this module however.
I'm hoping @dnc40085 and @marcelstoer in particular might be willing to lend a hand there.

Current API overview:
```lua
wifi.mode(wifi.NULLMODE || STATION || SOFTAP || STATIONAP)
wifi.getmode()
wifi.start()
wifi.stop()
wifi.on("sta_start" || "sta_stop" || "sta_connected" || "sta_disconnected" || "sta_authmode_chang" || "sta_got_ip" || "ap_start" || "ap_stop" || "ap_sta_connected" || "ap_sta_disconnected" || "ap_probe_req", function() ... end || nil)

wifi.sta.config({
  ssid="<ssid>",
  pwd="<password>",
  bssid="<bssid>",
  auto=1 || 0})
wifi.sta.connect()
wifi.sta.disconnect()
wifi.sta.getconfig() -- returns object that can be passed to wifi.sta.config()
wifi.sta.scan({
  -- filtering by sssid/bssid/channel/hidden not yet supported
}, function(ap_array) ... end) -- each found network has { ssid=, bssid=, channel=, auth=, rssi=, bandwidth= }

wifi.ap.config({
  ssid="<ssid>",
  pwd="<password>",
  auth=wifi.ap.AUTH_OPEN || AUTH_WEP || AUTH_WPA_PSK || AUTH_WPA2_PSK || AUTH_WPA_WPA2_PSK,
  channel= <valid_channel>,
  hidden=0 || 1,
  max=1..4 -- depends on IDF configuration too, possibly should not even expose this?
  beacon= <valid_beacon_interval>
})
```

Things to do:
- [ ] Document the module
- [ ] Power saving support
- [ ] Protocol/bandwidth/country/station-deauth/etc support

**Priority:** 
 - High for documentation
 - Medium for additional features

**Dependencies:**
None?

luismfonseca, 2016-12-02:
I've a minor suggestion on the `wifi.on` API.

Seeing that all options are prefixed with either "sta" or "ap", it could make more sense to have them on `wifi.sta` and `wifi.ap`.

So instead of:
`wifi.on("sta_start" || "sta_stop" || "sta_connected" || "sta_disconnected" || "sta_authmode_chang" || "sta_got_ip" || "ap_start" || "ap_stop" || "ap_sta_connected" || "ap_sta_disconnected" || "ap_probe_req", func)`

You could have:
`wifi.sta.on("start" || "stop" || "connected" || "disconnected" || "authmode_change" || "got_ip", func)`
`wifi.ap.on("start" || "stop" || "connected" || "disconnected" || "probe_req", func)`

jmattsson, 2016-12-03:
That sounds like a cleaner approach I must admit. It'll be a little bit more code to do it that way, but I think it's probably worth it. Thanks for the suggestion!

LDmitryL, 2017-05-16:
@jmattsson  Could you pay attention in #1969   and #1970 . It looks like there is a serious problem not allowing the use of ESP32. Or tell me who can understand this.

closed 2019-08-04 by stale[bot]