nodemcu/nodemcu-firmware #1616: ESP32: dhcp module

jmattsson, 2016-11-27:
**Rationale:**
The ESP32 can have up to three network interface - wireless client, wireless AP, and ethernet (yes, ethernet). The DHCP (client/server) functionality can be applied to these interfaces. Thus, rather than having the DHCP control live nested inside the `wifi` module, it should be made its own module.

I believe the IDF "does the right thing" by default, so explicit control is not a high priority.

- [ ] Implement DHCP client module that can be used on any (multiple) interface(s)
- [ ] Implement DHCP server module that can be used on any (multiple) interface(s)

**Priority:** Medium/Low

**Dependencies:**
  - Possibly IDF 1.0 release?

djphoenix, 2016-11-27:
Don't forget that ESP32 have bluetooth module, so (maybe) bluetooth networking may be implemented on it. Anyway I wait for IDF1.0 with (maybe complete) BT stack to just see "what it got".

jmattsson, 2016-11-27:
Hmm, I suppose IP over BlueTooth might be possible. We'll have to see what Espressif releases. The BT stacks (LE and Classic) are scheduled for release after IDF 1.0, probably ~3weeks from now last I heard.

igrr, 2016-12-15:
Thanks for the point about IP over BLE. We will consider this.

closed 2019-06-21 by stale[bot]