nodemcu/nodemcu-firmware #1617: ESP32: spi module

jmattsson, 2016-11-27:
**Rationale:**
We need to support the SPI bus(es). Hoping @devsaurus might sign on for this ;)

- [ ] Basic SPI bus support.
- [ ] SPI DMA support.

**Priority:**
  - High for basic support
  - Low for DMA version?

**Dependencies:**
  - IDF 1.0 released.

devsaurus, 2016-11-27:
I'd be on board, for sure 👍
Though the lack of ESP32 hardware will be blocking any heavy duty work for (quite) some time.

devsaurus, 2017-04-25:
I've started to sketch an API description in the original post. ATM this seems to evolve closely along the IDF concept, meaning:
- host/bus configuration with `spi.init(bus, bus_config)`
- add up to 3 devices to a bus with `spidev = spi.device(bus, dev_config)`
- run transactions on a device with `spidev:transaction()`

Transactions are synchronous by default in the basic implementation. Asynchronous operation can be added later by means of an optional callback (DMA is used by the driver in any case).

This is quite different from the Lua API on esp8266 and I tend to *not* be compatible with that but focus on exposing all of the hardware/driver functionality instead. I expect that backward compatibility can be retro-fitted with a Lua module up to a basic extent (to be proven).

Everybody feel free to feed back.

devsaurus, 2017-05-01:
Updated API to support SPI slave mode with latest esp-idf.

devsaurus, 2017-05-05:
Implementation started @ https://github.com/devsaurus/nodemcu-firmware/tree/esp32-spi.

devsaurus, 2017-05-14:
I've pushed initial SPI master support to `dev-esp32` branch (819284530e550d5343e6a0f9dd9bdd607c911299). Slave mode to follow.
NB there are 2 pending issues with the underlying IDF driver: Espressif/esp-idf#574 and Espressif/esp-idf#598.

devsaurus, 2019-06-10:
Closing this. SPI slave functionality should be handled as a new issue.

closed 2019-06-10 by devsaurus