nodemcu/nodemcu-firmware #1618: ESP32: I2S module

jmattsson, 2016-11-27:
**Rationale:**
The I2S peripheral seems like a very versatile thing, we totally should support it in a way that allows people to do weird and wonderful things with it!

- [ ] Basic I2S support
- [ ] I2S DMA support

**Priority:** Medium

**Dependencies:**
 - IDF 1.0 released?

devsaurus, 2016-11-27:
Agree, I2S enables a bunch of customized use cases beyond native audio.
Would be interesting to have an input path as well if the hardware supports that.

zelll, 2017-04-29:
I'm working on it. https://github.com/zelll/nodemcu-firmware/tree/dev-esp32-i2s
I tested it with a logic analyzer. Works, but there are still some issues. 
1. Latest esp-idf fixed some I2S issues, so i used i2s driver from latest esp-idf instead of [esp-idf @ fd3ef4c](https://github.com/espressif/esp-idf/tree/fd3ef4cdfe1ce747ef4757205f4bb797b099c9d9) . 
2. https://github.com/zelll/nodemcu-firmware/blob/dev-esp32-i2s/components/modules/i2s.c#L102
  If I use `ESP_INTR_FLAG_LEVEL1` here, only one I2S can be started. The second one will failed because interrupt allocation fails. `ESP_INTR_FLAG_LEVEL2`  works. I don't know what interrupt flag level means...
3. If I call `node.restart()` when `I2S 1` working ( without calling `i2s.stop()` ), the chip reboots, then if I call `i2s.start()` to start `I2S 1` again, the system will die (prints some unreadable chars and hangs up).
  It's weird that doing the same thing with `I2S 0` will not kill the system. And the story becomes more complex when combining `I2S 0` and `I2S 1`...
4. The code is not so clean as it calls IDF APIs directly, not `platform_xxx` functions.

jmattsson, 2017-04-30:
Great to hear you're working on it!

1. No worries, we're happy to keep tracking the latest IDF. I used to update it ~weekly, but I've been busy with other things lately.
1. I believe the levels are effectively the interrupt priority, as in if there are two outstanding interrupts, which one gets delivered first. Using level 2 *should* be fine in most if not all circumstances I think. @devsaurus you've been down in these layers more recently than I have, any comments?
1. That sounds like it might be a bug in the IDF or even the chip. Have you tried asking over on esp32.com or raising an issue on the IDF github repo? Is there some extra initialisation you can do in the module's init handler perhaps, as a workaround?
1. At this stage that's not a problem as far as I'm concerned; the ESP32 is the only one with the I2S hardware at this point. I'd rather have a functional I2S module in now, and worry about potential platform independence when/if we actually need it. Agile development and all that :)

devsaurus, 2017-04-30:
Great to see progress on this aspect, thanks @zelll!

2. I don't have a clear picture there to be honest and would not give any guidance here. So far I checked whether the ISR can run exclusively from IRAM (`ESP_INTR_FLAG_IRAM`), and eventually set the `ESP_INTR_FLAG_LOWMED` combo. You might get some enlightening thoughts from sdk's `intr_alloc.c`.
3. Happens also when you do an external reset? If not then this would hint towards the driver code missing to initialize hardware resources.
4. My view so far: Setting up an intermediate layer in platform makes sense when functionality is implemented using specific hardware/driver. E.g. on ESP32 the ws2812 protocol is implemented using the RMT driver while on ESP8266 it's on top of UART -> the protocol can be abstracted in platform. Same for 1-wire protocol on top of RMT for ESP32 but using GPIO for ESP8266. Even if I didn't move these into the platform layer, I would have placed the protocol code in a dedicated source file to separate it from the Lua bindings.
If you "just" expose driver functionality as a Lua module then there's not much benefit from an intermediate platform wrapper IMO. I plan so for the SPI master, followed this approach with I2C hardware master and slave, and will rework the sigma-delta driver at some point.

zelll, 2017-05-06:
2. The chip died when I was using a 921600 baudrate. After switching back to 115200, the chip just reboot in the test.
It works with no issue after a external reset. Only software reboot causes problem. I opened an issue a week ago: https://github.com/espressif/esp-idf/issues/558 . No response yet.

devsaurus, 2018-10-12:
I'll look into adding an I2S module next.

@zelll your code appears quite complete, so I might borrow from it :wink:

devsaurus, 2018-10-14:
An [i2S module](https://github.com/nodemcu/nodemcu-firmware/blob/dev-esp32/components/modules/i2s.c) is now available thanks to the foundation provided by @zelll :smile:.

closed 2018-10-14 by devsaurus