nodemcu/nodemcu-firmware #1620: ESP32: [wall]clock module

jmattsson, 2016-11-27:
**Rationale:**
The ESP32 finally has the capability of keeping wallclock/real time, even when deep sleeping. The IDF provides wallclock time access via the standard C interface `[gs]ettimeofday()`. This should be exposed in some user-friendly manner. If we're feeling particularly brave, we could even pull in timezone data... Or, we could just stick with UTC and save our sanity.

- [ ] Implement basic get/set time function.
- [ ] Implement `gmtime()` access for nice, easy (but UTC-only) calendar info.
- [ ] Implement timezone support and `localtime()` access.

**Priority:** Medium/Low

**Dependencies:**
None?

djphoenix, 2016-11-27:
I think this should be in both 8266 and 32 branches. About local times, I propose to just expose `[gs]ettimezone` to userspace, I think this is enough for most cases. TimeZone database is very huge for controllers.

igrr, 2016-12-15:
I was thinking that could add an option to bundle (limited amount of) timezone data with the firmware. Selection could be done in menuconfig, for instance. Oddly enough, I haven't found an existing way to convert entries from tzdata database into the [TZ variable format](https://www.gnu.org/software/libc/manual/html_node/TZ-Variable.html).

pjsg, 2016-12-15:
In my app, I have some lua code that handles the binary timezone files. In my case, I just upload the single required file (and call it "timezone") and then it just works.

LDmitryL, 2017-04-27:
@jmattsson There is progress in the development module?

jmattsson, 2017-04-27:
Alas, no progress on my end - work has me off on two different projects that aren't ESP32 related :(

HHHartmann, 2019-10-24:
Seems to be fixed already

PR #2561 (2018-11-26): ESP32: Add time module
Fixes #1620 

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ ] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

New time module for manipulating system time/ calendar and controlling SNTP client.
Haven't fully tested all the functions tested and working as intended:
* initntp
* settimezone
* getlocal
closed 2019-10-24 by HHHartmann