nodemcu/nodemcu-firmware #1621: ESP32: encrypted flash support

jmattsson, 2016-11-27:
**Rationale:**
The ESP32 supports running with full encryption of the flash. This is desireable for many reasons, including not storing WiFi (and other) credentials in plain-text in the flash, where they are easily retrievable. The encryption/decryption is all handled in hardware, and to my understanding does not cause any additional latency during execution.

The flash encryption does impact all regular read/write access to flash however, so we need to get the NodeMCU32 codebase "flash-encryption clean". While I'm not fully across the area yet, I believe we'll need to update the platform layer to use `mmap()` rather than explicit `SPIRead()/SPIWrite()`. This would go for our filesystem drivers as well, though hopefully we'll have at least moved over to the IDF-native SPIFFS, saving us that particular work.

Ideally I'd like to see NodeMCU32 use flash-encryption by default as I believe it should be possible to have it pretty much "invisible" to the user. The sooner we get to this state, the less risk and overhead when introducing new code which may touch the flash. It's much easier to get it fixed when it breaks on first use, than weeks down the track when someone gives that particular combination a go.

 - [ ] Ensure all flash access is encryption-"clean".
 - [ ] Enable flash encryption by default.

**Priority:** Medium/High

**Dependencies:**
 - IDF 1.0 released.

igrr, 2016-12-15:
I've looked into this a few times, and I don't think we can make SPIFFS compatible with flash encryption. Flash encryption happens in 32 byte blocks, but SPIFFS uses the property of flash memory that one can clear bits, one at a time. We don't have a viable alternative for a filesystem compatible with flash encryption yet.
If someone is into filesystems and is up for such a task, please let me know.

That being said, NVS is compatible with flash encryption, but it doesn't use it yet. This is on my list of things to do for the next version.

djphoenix, 2016-12-15:
@igrr why not? SPIFFS just calls storage read/erase/write methods, so we just need to wrap it to something that calls different ones depending on encryption enabled. Size of blocks are not significant - we may wrap it as we need.

jmattsson, 2016-12-15:
What about unencrypted flash with encrypted code? Do I understand things correctly that the en/decryption happens in hardware for mapped memory, but regular flash reads will return the raw bytes? Would that then make it possible to effectively use non-mapped areas of flash as plain flash storage? Or is there something that breaks that approach (does the initial encryption pass always process the full flash)?

igrr, 2016-12-15:
@djphoenix SPIFFS does indeed call read/write/erase methods, but it does make assumptions about the underlying storage. For example, after erasing a sector and writing, say, 0xfffffffe to some location, the assumption is that it is possible to write the same location once more with 0xffffffee, without erasing the sector first. Please check SPIFFS tech spec, this approach is referred to as ["nand way of writing"](https://github.com/pellepl/spiffs/blob/master/docs/TECH_SPEC#L218-L220). If we encrypt these writes, this assumption no longer holds true. Clever tricks used by SPIFFS to use individual bits as flags which can be toggled at any time are invalidated by flash encryption which transforms each write to more or less random set of bit flips.

One could of course detect this condition, erase the sector, and write it again from scratch, but I'm afraid this would invalidate much of the wear-leveling design of SPIFFS. I suppose @pellepl could give a better idea about how extensive is the use of "nand writes" in SPIFFS.

Second thing is that ESP32 flash encryption operates in 32 byte blocks. If SPIFFS happens to write, say, 10 bytes, and then another 10 bytes, this will not translate well to encrypted writes (you would need to erase the sector before doing the second 10 byte write).

@jmattsson Unencrypted mode is still possible: only cache reads are decrypted. Access to the SPI flash via SPI1 controller (i.e. SPIRead/SPIWrite) still works with plain data. `esp_partition_*` APIs will check `encrypted` flag of the partition and use plain reads and writes or encrypted ones depending on the flag. Encryption pass only processes those partitions which have `encrypted` flag set in the partition table.

rojer, 2017-01-16:
@igrr can you please clarify this passage in the NVM documentation:

>  Flash encryption can prevent some forms of modification:
> - replacing keys or values with arbitrary data
> - changing data types of values

what exactly does "replacing with arbitrary data" mean?

igrr, 2017-01-17:
When using NVS without flash encryption (that's what we have now in master), anyone can read our NVS, change keys or values at will, fix CRCs, and write back modified data.
With flash encryption in place, such arbitrary modifications are not possible. One can still create "Frankenstein" partition by replacing some encrypted NVS region with an encrypted region obtained at different point in time (but at the same address).

The support for flash encryption for NVS is mostly done, but I don't have time to merge it before 2.0, so it will land shortly after.

rojer, 2017-01-18:
@igrr ok, thanks.
in the mean time, i've been pondering encrypted FS support for mongoose-os and came up with an interim solution - encrypt file data and names, but leave spiffs structures plain-text. so, it will be possible to tell how many files are on the fs and their approximate size (to within 32 bytes), but not their names or content. i think this is a good enough solution. fs integrity can be further protected from tampering by protecting metadata structures with a checksum.
however, this requires one thing: being able to encrypt memory to memory. from what i can see, currently this is not possible - it seems that spi flash encryption block outputs the result only to SPI controller. which... kinda sucks. it means that in order to implement FS encryption with the same key as used for other flash encryption i'll need to have a scratch flash space which i use to write out encrypted data and then read it back to write out again immediately, this time to the spiffs area. this is ... less than ideal.
i even tried reading out the encryption data registers (`0x3ff5b000 - 0x3ff5b01c`) after encryption, hoping that encryption result may be available that way. no dice.

rojer, 2017-01-18:
so, to sum up: it would be ideal if the encryption peripheral provided the result in the registers after the operation is done. then the user would have flexibility with what to do with the result - either pipe it directly to the SPI controller or use it in some other way. i don't think this compromises security of the system in any way but it increases the flexibility. @igrr, wdyt?

pellepl, 2017-01-18:
Quick question: wouldn't a change of one plaimtext byte change the rest of
the ciphered file, or do you plan to encrypt individual 32byte blocks of
each file?

Regarding the encryption block it would be very weird if there isn't a ram
to ram possibility. I'm an ESP32 imbecile, but if one would
disassemble/debug the flash encryption routines one should see the setup.
Perhaps you've already done this though.

rojer, 2017-01-18:
@pellepl i plan to do what esp32 does - key tweaking based on offset. every 32 bytes the key is tweaked, so no, one byte change won't cascade through the entire file.

well, i also hoped so and [this comment](https://github.com/espressif/esp-idf/blob/0bbc60618bcd1cbe44c546f0c4e38a83f0597ad2/components/spi_flash/flash_ops.c#L275) raised my hopes, but having studied the internals of `SPI_Encrypt_Write` it seems that no, encryption result is never brought to ram. instead, what happens is SPI write source is switched to encryption block output, data is fed into the encryption peripheral (by `SPI_Prepare_Encrypt_Data`) and then a write is performed, which presumably takes its input directly from the encryption peripheral. as i said, i tried reading out the data registers used to feed the data in but the reads just return 0. bummer.

pellepl, 2017-01-18:
@rojer ok, that's a true bummer. What an .. interesting choice of architecture from espressifs side. Could it be that there are two hw cryptoblocks, one for flash and one for wifistuff? Perhaps they hardwired the "flash" crypto to prevent bus stalls or something like that.

In any case, shout if you need assistance with encrypting spiffs the way you mentioned. Keeping the structure and encrypting the data itself is the way to go as I see it. You might need another file length attribute embedded in the file data being plaintext file length, but other than that it should be pretty straightforward. Or perhaps you can pull it off without modifying spiffs. Let me know otherwise.

projectgus, 2017-01-19:
Something that may be helpful, soon esp-idf will add support for writing encrypted 16 byte aligned blocks with a performance penalty (AES operates on 16 byte blocks, so for a 16 byte write we can read & decrypt the other 16 bytes in the 32 byte row and then re-encrypt those, resulting in a write of the same 16 bytes of ciphertext.) So if it's possible to rework SPIFFS to write 16 byte aligned data only, this could work for this purpose (the blocks being encrypted were always going to need to be a multiple of 16 bytes, due to AES.)

rojer, 2017-01-22:
i have implemented file-level SPIFFS encryption. names and data are encrypted with a key derived from flash encryption key. i would _much_ prefer if i could encrypt and decrypt with the key it directly, but [in the mean time](https://github.com/cesanta/mongoose-os/blob/6be12a1fb9846bca29fbac97bb9dd950c920c23d/fw/platforms/esp32/src/esp32_fs_crypt.c#L16)...

igrr, 2017-05-02:
Small update from ESP-IDF side: we now have a wear levelling library in IDF which is compatible with flash encryption. One can use FATFS (which already has vfs bindings) on top of wear levelling library.

djphoenix, 2017-05-02:
@jmattsson @devsaurus does it means that we may replace SPIFFS support with FATFS+WL without flash restrictions?

devsaurus, 2017-05-02:
@djphoenix from a high-level perspective I'd say yes. Working through #1613 (and discarding SPIFFS) should get us IDF's vfs layer with FatFs support for internal SPI flash and external SD/MMC.

jmattsson, 2017-05-03:
Hmm. Interesting!

I could definitely see there being an argument for moving to FATFS. Not needing separate filesystems for SD vs internal flash should reduce the RAM requirements buffer-wise I believe. Plus we'd get full directory support as well, together with the encrypted flash. I'd say I'm in favour of switching.

Other than diverging (more) from the ESP8266 branch, are there any other downsides people can think of?

TerryE, 2017-05-03:
@jmattsson  FATFS is not nand flash friendly.  If has some very "hot" sectors, so switching from SPIFFS to FATFS will shorten the life of the chips if the apps write to the FS on anything other than an occasional basis.

devsaurus, 2017-05-03:
Patent issues with LFN & SFN support if we tie the flash to FAT? Though these aren't specific to our Nodemcu firmware. Wonder what Espressif's statement is here.

djphoenix, 2017-05-03:
@devsaurus about patent bypassing: http://www.geek.com/chips/clever-linux-folk-find-way-around-microsoft-fat-file-system-patent-824951/

TerryE, 2017-05-03:
Yup, the MS patent relates to the long name to PC-DOS 8.3 character short file names.  If you don't implement long to short name mapping, then there's no infringement.

jmattsson, 2017-05-04:
The one feature I can see possibly being impacted is preparing the filesystem image offline and flashing it onto the device. At least here at work that's a crucial feature to have.

@igrr Am I reading the code correctly in that it puts the WL cfg and state at the end of the partition? And if so, does that mean that it would be possible to build a `partition_size - WL_cfg_and_state` sized FAT image offline, and load that before first boot? Would that be compatible with the WL logic? (and if not, could it be made to?)

igrr, 2017-05-04:
Re LFN support and patents: currently we have LFN as an option which is disabled by default. We do plan to add a workaround similar to the one used in Linux (essentially by removing/adjusting part of dir_register function to ignore SFNs).

Regarding initialization of WL partition: yes, i think it should work like that, will confirm a bit later. Maybe we also need to initialize the state sectors as well, but that should be pretty straightforward. Note that currently WL partition has logical sector size of 4096 bytes. In the next version we plan to support smaller 512-byte logical sectors, mainly to reduce memory requirements in FATFS. In the process of adding support for smaller logical sectors there may be some WL format changes. We'll keep the initalization feature in mind so that it still works after these format changes.

TerryE, 2017-05-04:
@igrr, wear levelling abstraction at the virtual sector level is absolutely essential IMO for an embedded technology such as the esp8266 or esp32 since flash failure that can't be _fully_ accommodated by S/W means the ESP module is now a failed component and means swap out which has huge implications for through life maintenance and logistics. 

SPIFFS is slow but it primarily optimised to maximise nand flash life.  I can understand the desire to move to alternative nand flash compatible file system that has better performance and integrity, but surely the primary (and IMO sole) advantage of FATFS is its use for interchangeable FS,  IT is really obsolete for any other use.  So what advantages if any are there in moving to this for on an embedded nand flash?  

I am lost on this one.

igrr, 2017-05-04:
Note: I'm not arguing about advantages of FATFS over SPIFFS. Since this is a discussion of flash encryption support, I thought I'd mention this option (FATFS on top of wear_levelling library on top of spi_flash), because it works perfectly fine on top of encrypted flash, unlike SPIFFS. If flash encryption is not a requirement, SPIFFS may very well be a better option.

The existing wear_levelling library does provide abstraction layer on top of physical sectors, minimizing worst case erase cycle count for any particular sector.

For now, logical sectors are the same size as physical sectors — 4k bytes. We do understand that this is suboptimal and we plan to support 512b sector size in the next version.

djphoenix, 2017-05-04:
@igrr, as a side-topic - are wear-levelling library will (may) be available on ESP8266 SDK?

jmattsson, 2017-05-04:
Besides being in the DCF standard, the other thing FAT has going for it is that the codebase is very mature.

TerryE, 2017-05-04:
> Besides being in the DCF standard

Johnny, you've lost me on that one.  Why is this relevant to the FS used on a nand flash chip surface mounted on an ESP embedded module?

FS level encryption is architecturally in conflict with nand storage optimisation.  Reading from nand memory does not age the storage, neither does setting bits to 0 .  However setting _any_ bit in a device block to 1 requires erasure and rewrite of the block.  A characteristic of a low level encryption is that changing _any_ bit to 0 _or_ 1 in any encryption block (which invariably set roughly a ¼ of the bits in the block from 0 to 1) therefore requires block erasure.  

Hence the number of erases per logical file write (whcih will also update FS metadata could involve 10 × more erases than a nand flash optimised FS.  This means that at best an encrypted nand-based FS will has at best perhaps a tenth of the life on an unencrypted one and even with the best wear levelling algo, worse that that.

So IMO this is a concept that sounds good on paper but will kill the real usable life for any application that has even moderate volatile content.

jmattsson, 2017-05-04:
>>  Besides being in the DCF standard

>Johnny, you've lost me on that one. Why is this relevant to the FS used on a nand flash chip surface mounted on an ESP embedded module?

SD MMC card = nand flash, with wear leveling.
Internal ESP flash = nand flash, now with wear leveling. 

Why would (V)FAT be good enough for the first but not the second?

I see no reason why we couldn't support both FAT and SPIFFS for the internal flash on the ESP32. You've lost me on why having options available is a bad thing :)

TerryE, 2017-05-05:
Johny, I use RPis which use an SD MMC as their system device, and like many others have found that even top quality cards, they can die suddenly and unexpectedly after a year or two.  So you have to keep spairs and back up off-device. I now use a usb-attached 2.5" HDD for the main FSs and only have the boot and config on the SD.  At least you can swap out the SD card on a RPi.  When you get the same sort of failure on an ESP module for most users it is bricked. If the embedded system doesn't use plug in replaceable modules and the spairs are readily available then the system is bricked.

So yes, so long. as the user can set up two FS on the device: one low volatility encrypted FS using the FAT and one SPIFFS if a higher volatility data is needed, then as you say it's an option. 

However developers are going to notice that writing to FAT is faster, so many might be tempted to use FAT for volatile storage for performance reasons, and enough end users / consumers are going to find their devices bricking after  6 months to 2 years, to give the ESPs a reputation for bad reliability -- at least that a bad case scenario of an outcome here. 😳

closed 2019-06-21 by stale[bot]