nodemcu/nodemcu-firmware #1623: ESP32: cloud builder support

jmattsson, 2016-11-27:
**Rationale:**
The NodeMCU cloud build facility has been extremely useful, especially since the firmware grew beyond a one-size-includes-all. While the NodeMCU32 firmware currently could include everything, there are numerous settings available at compile time that a user might want to tweak (in particular, log verbosity).

With the IDF (and thus, NodeMCU32) based on KConfig, I hope there's a possibility of extending the cloud builder to also handle NodeMCU32. I've been meaning to see if there are any existing web frontends to KConfig, but time has been (and still is) in short supply.

It might be that we'll have to provide a couple of "baseline" configs for the nitty-gritty stuff so users don't have to worry about those bits, but on the whole I'm hoping it can be possible to integrate nicely with the KConfig files.

@marcelstoer, you're the expert in this area by far, so the direction of this is largely going to be what you think is best.

**Priority**: Medium/High

**Dependencies:**
None.

marcelstoer, 2018-11-26:
As of today we're one step closer. At least building with Docker is now supported.

marcelstoer, 2019-12-05:
Sigh, it almost physically hurts to see that this was created some three years ago - and still open 😭 I wish a fairy would just magically make the Docker scripts available in https://github.com/marcelstoer/nodemcu-custom-build/

marcelstoer, 2020-11-29:
Available since February 2020: https://twitter.com/frightanic/status/1226991992917385224

closed 2020-11-29 by marcelstoer