nodemcu/nodemcu-firmware #1624: Build fails on OS X

djphoenix, 2016-11-27:
### Expected behavior
Success build

### Actual behavior
```
xtensa-lx106-elf-gcc <...>  -o .output/eagle/debug/obj/dbg_printf.o -c dbg_printf.c
dbg_printf.c:147:1: error: conflicting types for 'kprintn'
 kprintn(void (*put)(const char), unsigned long ul, int base, int width, char padchar)
 ^
dbg_printf.c:47:13: note: previous declaration of 'kprintn' was here
 static void kprintn (void (*)(const char), uint32_t, int, int, char);
             ^
dbg_printf.c:47:13: error: 'kprintn' used but never defined [-Werror]
```

### NodeMCU version
`dev` branch (commit c54bc05ba61fe55f0dccc1a1506791ba41f1d31b)

```
$ xtensa-lx106-elf-gcc -v
Using built-in specs.
COLLECT_GCC=xtensa-lx106-elf-gcc
COLLECT_LTO_WRAPPER=/Users/phoenix/Dev/esp-open-sdk/xtensa-lx106-elf/libexec/gcc/xtensa-lx106-elf/4.8.5/lto-wrapper
Target: xtensa-lx106-elf
Configured with: /Users/phoenix/Dev/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/configure --build=x86_64-build_apple-darwin16.0.0 --host=x86_64-build_apple-darwin16.0.0 --target=xtensa-lx106-elf --prefix=/Users/phoenix/Dev/esp-open-sdk/xtensa-lx106-elf --with-local-prefix=/Users/phoenix/Dev/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot --with-sysroot=/Users/phoenix/Dev/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG crosstool-ng-1.22.0-55-gecfc19a' --disable-__cxa_atexit --enable-cxx-flags='-fno-exceptions -fno-rtti' --with-gmp=/Users/phoenix/Dev/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-mpfr=/Users/phoenix/Dev/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-mpc=/Users/phoenix/Dev/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-isl=/Users/phoenix/Dev/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-cloog=/Users/phoenix/Dev/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-libelf=/Users/phoenix/Dev/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-libgomp --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libquadmath-support --disable-nls --disable-multilib --enable-languages=c,c++
Thread model: single
gcc version 4.8.5 (crosstool-NG crosstool-ng-1.22.0-55-gecfc19a) 
```

### Patch

```diff
diff --git a/app/libc/dbg_printf.c b/app/libc/dbg_printf.c
index 92f3668..02ba01a 100644
--- a/app/libc/dbg_printf.c
+++ b/app/libc/dbg_printf.c
@@ -144,7 +144,7 @@ reswitch:	switch (ch = *fmt++) {
 }
 
 static void
-kprintn(void (*put)(const char), unsigned long ul, int base, int width, char padchar)
+kprintn(void (*put)(const char), uint32_t ul, int base, int width, char padchar)
 {
 					/* hold a long in base 8 */
 	char *p, buf[(sizeof(long) * 8 / 3) + 2];
```

djphoenix, 2016-11-27:
The reason for this issue is to warn developers about "shorthand types is not same that expanded". Of course I will fix it in sdk200 branch.

marcelstoer, 2016-11-27:
Just for clarification, as to not confuse casual observers...`sdk200` is _your_ branch (i.e. in your forked repository) and not ours 😉

djphoenix, 2016-11-27:
Yep. But regular sdk update *should not* break build in normal cases. ;)

closed 2016-12-11 by marcelstoer