nodemcu/nodemcu-firmware #1630: Cant connect with STATION mode in my home router

rsabin, 2016-12-02:
### Expected behavior
I am trying to connect to my router. Simple and easy.

### Actual behavior
I can't connect and don't know the reason.
**The issue only happen with my _home router_. If I create a _Personal Hotspot_ with my iPhone, for example, it works fine.**
I should add that my router is very simple, no blacklists, everyone can connect providing the correct ssid and pwd. All my computers, TVs and cellphones are connecting ok. The problem is the nodemcu devkit only.

### Test code
Nothing exceptional, I am just calling wifi.sta.config() with correct credentials.
This is my program:
```Lua
wifi.setmode(wifi.STATION)
wifi.sta.config("my ssid", "my pwd")

cont=1
tmr.alarm(1, 2000, 1, function()
   if wifi.sta.getip()==nil then
      print("Wait for IP "..cont.."--> "..wifi.sta.status())
      cont=cont+1
      if cont>20 then tmr.stop(1) end
   else
      print("New IP address is "..wifi.sta.getip())
      tmr.stop(1)
   end
end)
```
### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: master
	commit: ec265a6c21db22640795f190bdcb8a4f014cdced
	SSL: false
	modules: file,gpio,net,node,tmr,uart,wifi
 build 	built on: 2016-12-01 22:57
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

### Hardware
It's an NodeMCU devkit.

### More info
If this help, I made a custom build with debug on and this is the results on ESPlorer:
wifi_event_monitor_handle_event_cb is called
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)

Also, I can try any code you send to help testing.

jmattsson, 2016-12-02:
Tagging @dnc40085 as the WiFi expert :)

dnc40085, 2016-12-02:
Please add the code below to your init.lua, make sure it is before the call to wifi.sta.config().
Then, check the disconnect reason. 

if `reason: 201` is returned, please verify that the ssid being passed to `wifi.sta.config()` is the same as the one in the table returned by wifi.sta.getap().
Otherwise, please post the reason here.

```lua 
wifi.eventmon.register(wifi.eventmon.STA_DISCONNECTED, function(T) 
 print("\n\tSTA - DISCONNECTED".."\n\tSSID: "..T.SSID.."\n\tBSSID: "..
 T.BSSID.."\n\treason: "..T.reason.."\n")
 end)
```

rsabin, 2016-12-02:
Hi @dnc40085 , first of all: thanks for the help.

This is the disconnect reason:
**wifi.eventmon.reason.ASSOC_FAIL 	203**

dnc40085, 2016-12-02:
@rsabin You're welcome!

According to [supportforums.cisco.com](https://supportforums.cisco.com/document/141136/80211-association-status-80211-deauth-reason-code) the error is defined as:

 code | 802.11 definition | Explanation 
----------------|-------|-----
 203(0xCB) | Invalid Parameters. The request has not been successful as one or more TSPEC parameters in the request have invalid values. A TSPEC SHALL be present in the response as a suggestion. Not defined in IEEE, defined in CCXv4 | This happens in cases such as PHY rate mismatch. If the TSRS IE contains a phy rate not supported by the controller, for example. Other examples include sending a TSPEC with bad parameters, such as sending a date rate of 85K for a narrowband TSPEC.

what is the make / model of your router?

rsabin, 2016-12-03:
Hi, its a "thing"called **Power Box**.
It came configured by my provider, I can't edit many things.
_Vendor: **PACE** Model: **V5471** Software Version: B14103-GVT-RC2-103140_
[image](https://http2.mlstatic.com/modem-roteador-gvt-powerbox-pace-v5471-002-D_NQ_NP_701215-MLB25163905514_112016-F.jpg)
I tried to find a user manual for you but no success. Even at vendor website.

Some of the allowed configs are (bold are my current choices):
> Private Wi-Fi Network: **on**|off
> SSID Broadcast: **on**|off
> Security Mode: WPA2|**WAP+WPA2**
> Operation Mode: 802.11b|802.11g|802.11n|802.11b+g|**802.11g+n**|802.11b+g+n
> Channel: **Auto**|1..13
> Channel Bandwidth: Auto|20|**40**
> Transmit Power: 1%..**100**%
> WMM: **on**|off
> Power Saving - WMM: **on**|off
> Frequency Band: **2.4Ghz**

Shoul I add that my current ssid and pwd are using just letters an numbers. I read somewhere that special characters may cause trouble.

dnc40085, 2016-12-03:
>Channel Bandwidth: Auto|20|_40_

try changing `channel bandwidth` to `auto`

rsabin, 2016-12-03:
No use... same thing.

dnc40085, 2016-12-03:
Unfortunately my knowledge of the inner workings of WiFi is a work in progress.
I would suggest capturing some packets with Wireshark, but I wouldn't even know where to start on deciphering the output... maybe @pjsg or @devsaurus might be able to help.

devsaurus, 2016-12-03:
I'd guess that you'd need to capture the association phase between ESP and router with a separate PC in wifi promiscuous/monitoring mode. Can probably be done with Wireshark, but I don't have experience with this.

pjsg, 2016-12-03:
The problem is that you need to get a wifi device that can run in monitor mode. If you have linux or OSX then there is a chance. Run the latest version of wireshark and capture in monitor mode. I'm on a macbook pro and I could do it -- but you have to use the capture/options window and scroll over to the right where you will find the monitor mode checkbox. This will show you all the 802.11 frames....

rsabin, 2016-12-03:
Well, I was able to capture some wifi trafic using Wireshark and trial version of acrylic wifi... I just dont know where to start looking for answers.
If I upload the capture files, someone mind helping me a little more?
This is a example, just after reseting the NodeMCU (init.lua tried to connect).
[image]
SabinNET is my ssid...

pjsg, 2016-12-03:
Can you attach the pcap file -- so I can look at the frames....

rsabin, 2016-12-03:
[cap.zip](https://github.com/nodemcu/nodemcu-firmware/files/629161/cap.zip)
Thank you.

pjsg, 2016-12-03:
I compared against my setup, and the only difference that I could see was that the Probe Response packet from your router was large (391 bytes). My AP's response varies by requesting device -- and is never more than 258 bytes. 

I wonder if there is a fixed size buffer in the espressif code and your router sends an overlong response which gets dropped.

simonvanderveldt, 2017-01-16:
I'm experiencing the same issue, can't connect to my access point (I have actually been able to connect to it some time ago, but don't really know/remember what's changed in between) but can connect to my phone's access point. I'm using an ASUS Asus WL-500gP using Tomato firmware. Because of this it does have a fair amount of toggles/options, maybe some of them is causing issues? This is the list I have as advanced options
[image]

When adding the logging/debugging code from https://github.com/nodemcu/nodemcu-firmware/issues/1630#issuecomment-264449777 it outputs the same issue, being "reason: 203".

Is the only thing we can do get a pcap and compare between connecting to my phone's AP and my main AP?

andrebstv, 2017-02-24:
Anyone with news on the topic?

rsabin, 2017-02-24:
I got a new router from my ISP. It is a Sagemcom Fast 2764.
It is working fine for now.

@simonvanderveldt, my only guess is that your router got a new firmware or update and then you have the same problem.

Until someone in the nodemcu-firmware team with the same issue handle the problem, I suggest you to work with a wifi repeater.

andrebstv, 2017-02-24:
@rsabin I'm having the same problem with the Arduino Framework for esp8266. It's not a nodemcu firmware problem.

I believe, as @pjsg said, it might be a problem on the espressif SDK.

Anyone would like a hand to tackle the problem?

andrebstv, 2017-02-24:
@rsabin check this:

"I've seen some AP where the probe response became too long (it was in 11n-mode with WPS enabled) for the client to associate but when disabling WPS solved it."

Could you test it?

Source: https://www.cwnp.com/forums/posts?postNum=288568

rsabin, 2017-02-24:
Sorry @andrebstv, but my ISP take the old router and give me this new one... I can't test anymore.

But this is a good try for people still with errors.
My old router indeed was 802.11g+n and WPS enabled.
I tried to change the .11 sometimes but never tried to disable WPS.

simonvanderveldt, 2017-02-24:
Tomato doesn't support WPS, so that's definitely not the cause of problems for me

andrebstv, 2017-02-24:
@simonvanderveldt have you tried disabling it and the 802.11n ?

simonvanderveldt, 2017-02-24:
> @simonvanderveldt have you tried disabling it and the 802.11n ?

@andrebstv sorry, I don't understand what you mean with the above message.
I'm using an older router that doesn't support 802.11n and it's running Tomato which doesn't support WPS.

andrebstv, 2017-02-24:
Disable 802.11n and/or allow only allow 802.11g. Then also disable WPS

yuraj11, 2017-03-03:
Same problem here: http://bbs.espressif.com/viewtopic.php?t=2140

simonvanderveldt, 2017-03-03:
@yuraj11 you never hear back from espressif? Maybe you can ping them again on that forum?

closed 2017-04-30 by marcelstoer

lordbbk, 2018-06-05:
@simonvanderveldt, I have a ZTE router and encountered the same problem. I could however circumvent the problem such a way that I enabled also 802.11b mode (previously I have set 802.11g+802.11n only, and esp could not connect). Now it could connect perfectly -- despite the fact that I instructed in the sketch to use 802.11n with `WiFi.setPhyMode(WIFI_PHY_MODE_11N);` but it seems that it wants to connect only with speed `b`.