nodemcu/nodemcu-firmware #1633: crypto.fhash() causes reboot when hash an empty file

kicer, 2016-12-02:
According to the documentation crypto.fhash() should:

> Compute a cryptographic hash of a a file.

Returns

> A binary string containing the message digest. To obtain the textual version (ASCII hex characters), please use crypto.toHex()

But when I try hash an empty file, will causes reboot:

```
> file.open("empty.lua", "w+")   --create empty file
> file.close()
> print(crypto.toHex(crypto.fhash("MD5", 'empty.lua')))

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 24684, room 16 
tail 12
chksum 0x46
ho 0 tail 12 room 4
load 0x3ffe8000, len 2204, room 12 
tail 0
chksum 0xe6
load 0x3ffe889c, len 8, room 8 
tail 0
chksum 0x52
csum 0x52

NodeMCU 1.5.4.1 build unspecified powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
lua: cannot open init.lua
```

Version used:
branch: dev
commit: 11ded3f, **the #1592 has Fixed**
modules: _default_+**crypto**

kicer, 2016-12-08:
@pjsg, After #1647 patched, fhash() empty file works as expected. Thanks~

```
> print(crypto.toHex(crypto.fhash("MD5", 'empty.lua')))
d41d8cd98f00b204e9800998ecf8427e
> print(crypto.toHex(crypto.fhash("SHA1", 'empty.lua')))
da39a3ee5e6b4b0d3255bfef95601890afd80709
> print(crypto.toHex(crypto.fhash("SHA256", 'empty.lua')))
e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855
> print(crypto.toHex(crypto.fhash("SHA384", 'empty.lua')))
38b060a751ac96384cd9327eb1b1e36a21fdb71114be07434c0cc7bf63f6e1da274edebfe76f65fbd51ad2f14898b95b
> print(crypto.toHex(crypto.fhash("SHA512", 'empty.lua')))
cf83e1357eefb8bdf1542850d66d8007d620e4050b5715dc83f4a921d36ce9ce47d0d13c5d85f2b0ff8318d2877eec2f63b931bd47417a81a538327af927da3e
```

marcelstoer, 2016-12-12:
Johny why did you close this? ATM #1647 isn't merged yet and we seem to have settled for a policy of closing issues only once the PR which fixes them is merged?

jmattsson, 2016-12-12:
I was sure it was merged! ./brainfart, sorry!

closed 2016-12-25 by marcelstoer