nodemcu/nodemcu-firmware #1634: Wifi settings not saved to flash on 16MB Wemos Mini Pro

devsaurus, 2016-12-02:
### Expected behavior
Wifi settings are saved permanently to flash as documented in [`wifi.setmode()`](http://nodemcu.readthedocs.io/en/dev/en/modules/wifi/#wifisetmode) or [`wifi.sta.config()`](http://nodemcu.readthedocs.io/en/dev/en/modules/wifi/#wifistaconfig).

### Actual behavior
Commands which require to save the configuration to flash do not work / are not effective. After reset always `wifi.SOFTAP` mode is active.

Executing the configuration commands without saving to flash appears yield a working volatile configuration. It needs to be re-applied after the next reset.

### Test code
Sample code that runs flawlessly on a newly flashed NodeMCU V1.0 board:
```lua
print(wifi.getmode())
2

print(wifi.setmode(wifi.STATION, true))
1

print(wifi.getmode())
1

station_cfg={}
station_cfg.ssid="NODE-AABBCC"
station_cfg.pwd="password"
station_cfg.save=true
print(wifi.sta.config(station_cfg))
true

print(wifi.sta.status())
5
```

On the Wemos Mini Pro:

```lua
print(wifi.getmode())
2

print(wifi.setmode(wifi.STATION, true))
2

print(wifi.getmode())
2

station_cfg={}
station_cfg.ssid="PirateBox"
station_cfg.pwd="1rn2m+k3r4n"
station_cfg.save=true
print(wifi.sta.config(station_cfg))
false
```

### NodeMCU version
Current `dev` branch.

### Hardware
OK on NodeMCU V1.0 board (4 MByte flash).
NOK on Wemos Mini Pro (16 MByte flash).

dnc40085, 2016-12-02:
I'm pretty sure the sdk uses the last 3 sectors(0,1=data, 2=flag) of the flash to store it's parameters via the functions(`system_param_save_with_protect` and `system_param_load`).

Only thing I can think of to do... 
set a few parameters that would save to flash (wifi.sta.config or wifi.setmode)
dump the last 3 sectors from 4mb esp and 16mb esp
compare the two and make sure the data is in the right place
if not, ????, maybe something is wrong with `user_rf_cal_sector_set` in `user_main.c` or something else.

devsaurus, 2016-12-03:
Good point to investigate the flash contents of the two. Will try that next.
Is there any doc/ref of the configuration sectors available?

dnc40085, 2016-12-03:
None that I've been able to find.

devsaurus, 2016-12-04:
> Good point to investigate the flash contents of the two.

Got stuck here - ROM and SDK are obviously not able to access flash beyond the 4&nbsp;MByte border. A lot of information is available from [ESP8266 16MB Flash Handling](http://www.packom.org/esp8266/16mb/flash/eeprom/2016/10/14/esp8266-16mbyte-flash_handling.html).

Thus my view on this is that the >4&nbsp;MByte modules won't work with the current memory layout of the firmware:
```
  0x00000
     firmware
  0xWWWWW
     padding
  0xXXXXX
     SPIFFS
  0xYYYYY
     rfcal sector
  0xZZZZZ
     4 SDK sectors
  End of Flash
```

The 4+1 sectors for SDK are abutted at the end of the flash, and I expect that this is a problem if addresses `0xYYYYY` and `0xZZZZZ` are beyond the 4&nbsp;MByte.

A solution would be to freeze the SDK flash size to let's say 1&nbsp;MByte and grow SPIFFS after that:
```
  0x00000
     firmware
  0xWWWWW
     padding
  0xYYYYY
     rfcal sector
  0xZZZZZ
     4 SDK sectors
  0xXXXXX
     SPIFFS
  End of Flash
```

Here the 4+1 SDK sectors are below 4&nbsp;MByte while access to SPIFFS above this border is already possible from the firmware.

Will give it a try soon.

PR #1646 (2016-12-26): Rework support for >4MB modules: place SPIFFS after SDK data
Fixes #1634.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

The default flash memory map strategy of the firmware doesn't scale beyond 4&nbsp;Mbyte / 32&nbsp;Mbit modules since SDK and ROM cannot access flash in the upper [... cut]
closed 2016-12-26 by marcelstoer