nodemcu/nodemcu-firmware #1635: Crash after sntp.sync

vsky279, 2016-12-03:
### Actual behavior
ESP8266 ESP-01 crashes when `net.dns.resolve`, `sntp.sync`, `rtctime.get` and `rtctime.epoch2cal` are called in this order.
The behavior was tested on two ESP8266 ESP-01.

### Test code
The following minimal code causes the crash when the code is called. When modified slightly, e.g. `net.dns.resolve` omitted and fixed ip address is used no crash occurs.
The issue usually occurs after software reset (`node.restart`). Sometimes it does not occur when the code is ran after the ESP is powered on.

```Lua
ts2gmt=function(ts)
    local tm = rtctime.epoch2cal(ts)
    return {tm["year"], tm["mon"], tm["day"], tm["hour"], tm["min"], tm["sec"]}
end

format=function()
    local ts=rtctime.get()
    local dt=ts2gmt(ts)
    return string.format("%04u/%01u/%01u %02u:%02u:%02u",unpack(dt))
end
    
syncdns=function()
    local server="0.cz.pool.ntp.org"
    net.dns.resolve(server, function(sk, ip)
            sntp.sync(ip,
                function (ts)
                    print("Time updated: ", format())
                end)
    end)
end

syncdns()
```
The crash occurs somewhere after `rtctime.get()`. When SNTP debugging is turned on it can be seen that `sntp.sync` finishes correctly.
The output before restart occurs:
```Lua
> 3ffefd08 already freed
```

### NodeMCU version
Current dev version (https://github.com/nodemcu/nodemcu-firmware/pull/1629/commits/211dfa256a26df40dee567a04c9d9169fc1c5f3c) - integer version.
These modules are included: `file`, `gpio`, `net`, `mode`, `rtctime`, `sntp`, `tmr`, `uart`, `wifi`.

### Hardware
ESP8266 ESP-01

pjsg, 2016-12-03:
Can you try building dev with https://github.com/nodemcu/nodemcu-firmware/pull/1596 merged in? I think that it solves your problem. At least it doesn't appear to break for me (though I have other changes in my version as well).

vsky279, 2016-12-04:
Great! With the fix I was not able to reproduce the crash.
When do you intend to merge?

closed 2016-12-04 by vsky279